project-chip/connectedhomeip #22210: StartUpColorTemperatureMireds does not get triggered on DUT boot

innovation-matters-iot, 2022-08-27:
#### Problem
If the DUT is rebooted no value is sent to restore the old CT.

Reading the value of StartUpColorTemperatureMireds from chip-tool results in:
[read_startup_ct.log](https://github.com/project-chip/connectedhomeip/files/9437595/read_startup_ct.log)

Follow up issue to: [21855](https://github.com/project-chip/connectedhomeip/issues/21855)

bzbarsky-apple, 2022-08-27:
@innovation-matters-iot what does the actual device configuration look like?  In particular, is the StartUpColorTemperatureMireds attribute set to RAM or NVM in the ZAP configuration?

And what are the exact commands being executed that demonstrate the problem?

innovation-matters-iot, 2022-08-29:
@bzbarsky-apple It is normally set to NVM, I just tested it with RAM but the result is the same - The DUT boots and the CT is not received in the devicecallbacks.

The command to demonstrate the problem of reading the StartUpColorTemperatureMireds from the DUT is:

- ./chip-tool colorcontrol read start-up-color-temperature-mireds 1 1

bzbarsky-apple, 2022-08-29:
@innovation-matters-iot So just to make sure I understand, this is the sequence of operations:

1. StartUpColorTemperatureMireds is set to NVM storage (good).
2. The client issues a write to StartUpColorTemperatureMireds to set it to a non-null value (say 25).
3. The server is rebooted.

At this point, reading from the server what is the value of StartUpColorTemperatureMireds?  And what is the value of ColorTemperatureMireds?  I would expect both to be 25, and that is in fact what I see with all-clusters-app as the server, running on a Mac...

closed 2023-03-23 by stale[bot]