project-chip/connectedhomeip #22217: [TC-DGWIFI-1.1] AcceptedCommandList should be 0 if DUT does not support ERRCNT

pankore, 2022-08-29:
#### Problem
- TC-DGWIFI-1.1
- FeatureMap reads 0
- AcceptedCommandList shows 1 entry
- AcceptedCommandList should be 0 if DUT does not support ERRCNT

[1661690037.104771][3049:3054] CHIP:TOO:   AcceptedCommandList: 1 entries
[1661690037.104849][3049:3054] CHIP:TOO:     [1]: 0

[1661733544.606882][2286:2291] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0036 Attribute 0x0000_FFFC DataVersion: 150492557
[1661733544.609771][2286:2291] CHIP:TOO:   FeatureMap: 0

[1661733554.618029][2292:2297] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0036 Attribute 0x0000_FFFB DataVersion: 150492557
[1661733554.618102][2292:2297] CHIP:TOO:   AttributeList: 10 entries 
[1661733554.618132][2292:2297] CHIP:TOO:     [1]: 0 
[1661733554.618154][2292:2297] CHIP:TOO:     [2]: 1 
[1661733554.618176][2292:2297] CHIP:TOO:     [3]: 2 
[1661733554.618197][2292:2297] CHIP:TOO:     [4]: 3 
[1661733554.618231][2292:2297] CHIP:TOO:     [5]: 4 
[1661733554.618264][2292:2297] CHIP:TOO:     [6]: 65528 
[1661733554.618285][2292:2297] CHIP:TOO:     [7]: 65529 
[1661733554.618325][2292:2297] CHIP:TOO:     [8]: 65531 
[1661733554.618348][2292:2297] CHIP:TOO:     [9]: 65532 
[1661733554.618379][2292:2297] CHIP:TOO:     [10]: 65533

andy31415, 2022-08-31:
@pankore please expand the test description:

please link to TC-DGWIFI-1.1 - many developers do not directly handle cert cases and are not familiar with every step. Specifically what FeatureMap is this refferring to as being 0?

Also add bug report information: 
  - what DUT was used
  - what SHA of the SDK is this based on?
  - is this reproducible on developer machines (e.g. chip-tool and light on linux or mac or similar)? 
  
  This will affect our prioritization (platform vs sdk bug).

andy31415, 2022-08-31:
Looking at test yaml file, DGWIFI seems to be "WiFi Network Diagnostics": https://github.com/project-chip/connectedhomeip/blob/master/src/app/tests/suites/certification/Test_TC_DGWIFI_1_1.yaml

pankore, 2022-09-01:
Platform issue, resolved.

closed 2022-09-01 by pankore