project-chip/connectedhomeip #22219: How we can enable/disable features using run_zaptool?

jadhavrohit924, 2022-08-29:
#### Problem
* How we can enable/disable features using run_zaptool.sh?
* How we can get details of features added in clusters using run_zaptool.sh?

bzbarsky-apple, 2022-08-29:
> How we can enable/disable features using run_zaptool.sh?

By setting the default value of the FeatureMap attribute for the relevant cluster and then enabling/disabling the relevant attributes/commands for the relevant cluster.

There is, so far, no support for any sort of smart mapping of feature flags to attributes/commands in ZAP.  Please file isses on https://github.com/project-chip/zap/ as needed.

> How we can get details of features added in clusters using run_zaptool.sh?

I'm not sure what this is asking.

jadhavrohit924, 2022-08-30:
Ok, Thanks.
Second question was about attributes and commands.

bzbarsky-apple, 2022-08-30:
@jadhavrohit924 What is the specific question about them?  How to tell whether a given attribute or command is enabled for a given cluster instance?

jadhavrohit924, 2022-08-30:
I want to create API's of attributes and commands of features using these xmls. Xml only tells up about which cluster have wich features but not the details like attributes and commands?

closed 2022-08-30 by jadhavrohit924