project-chip/connectedhomeip #22226: [TC-TSTAT-2.2] - Results CONSTRAINT_ERROR instead Success

sowmyassp, 2022-08-29:
**Issue** : Results CONSTRAINT_ERROR instead Success when we tried to write occupied-cooling-setpoint 

**CommitID** - fa8db560907713c555b06120658034e8c6ccea1a  (**SVE-2 branch**)

**Expected behaviour** : DUT expects Success 

**Actual behaviour** : DUT results CONSTRAINT_ERROR

**App used** - thermostat-app

**Steps to reproduce :**

1. DUT side :

sudo rm -rf /tmp/chip_*
sudo ./chip-all-clusters-app 

2. TH side

sudo rm -rf /tmp/chip_*
./chip-tool pairing ethernet 1 20202021 3840 fe80::e65f:1ff:fe0f:2753  5540
./chip-tool thermostat write occupied-cooling-setpoint -30  1 1

**PFA Logs:**
[TSTAT-2.2_cntrl.txt](https://github.com/project-chip/connectedhomeip/files/9443786/TSTAT-2.2_cntrl.txt)

bzbarsky-apple, 2022-08-29:
@chrisbeach59 please take a look?

chrisbeach59, 2022-08-29:
This is expected behavior.

sowmyassp, 2022-08-30:
@chrisbeach59 
Is the verification command correct here https://docs.google.com/spreadsheets/d/1h4d4J90KAxFqUL_cXZHyM3pHjHCEXzFITkQhN2KMqDY/edit#gid=1154794263&range=E130:G130 ??

chrisbeach59, 2022-08-30:
@sowmyassp Interesting that it shows when setting the cooling setpoint to -30 that they show constraint error.  Which would expect.  So the short answer is that the VS step is quite wrong.   I would have expected to the first part of the step to have been
thermostat write occupied-cooling-setpoint 2600 1 1  

I would recommend that thermostat cluster be tested with UI Automated testing.  It is critical to make sure the xml file matched the configuration of the thermostat cluster by ZAP

cjandhyala, 2022-08-31:
closing the issue per Chris's comment

closed 2022-08-31 by cjandhyala