project-chip/connectedhomeip #22227: library "libSetupPayloadParser.so" not found

mohitjadav86, 2022-08-29:
#### Problem
i am able to run the sample application using android studio but facing some issue **library "libSetupPayloadParser.so" not found**
and can't find any solution

Here is the stack-trace/error i am facing

Process: com.google.chip.chiptool, PID: 24464
    java.lang.UnsatisfiedLinkError: dlopen failed: library "libSetupPayloadParser.so" not found
        at java.lang.Runtime.loadLibrary0(Runtime.java:1077)
        at java.lang.Runtime.loadLibrary0(Runtime.java:998)
        at java.lang.System.loadLibrary(System.java:1656)
        at chip.setuppayload.SetupPayloadParser.<clinit>(SetupPayloadParser.java:31)
        at com.google.chip.chiptool.setuppayloadscanner.BarcodeFragment.handleInputQrCode(BarcodeFragment.kt:145)
        at com.google.chip.chiptool.setuppayloadscanner.BarcodeFragment.initializeBarcodeDetectorAndCamera$lambda-4(BarcodeFragment.kt:124)
        at com.google.chip.chiptool.setuppayloadscanner.BarcodeFragment.lambda$EBnjAD5Q2aHtnpJoXkWPUVvyiaY(Unknown Source:0)
        at com.google.chip.chiptool.setuppayloadscanner.-$$Lambda$BarcodeFragment$EBnjAD5Q2aHtnpJoXkWPUVvyiaY.onClick(Unknown Source:2)
        at android.view.View.performClick(View.java:7441)
        at android.view.View.performClickInternal(View.java:7418)
        at android.view.View.access$3700(View.java:835)
        at android.view.View$PerformClick.run(View.java:28676)
        at android.os.Handler.handleCallback(Handler.java:938)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loopOnce(Looper.java:201)
        at android.os.Looper.loop(Looper.java:288)
        at android.app.ActivityThread.main(ActivityThread.java:7839)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1003)

nicelyjust, 2022-08-29:
did you building *.so successfully? if did ,Please check if the arch matches

mohitjadav86, 2022-08-29:
no i am not able to build i have comment on it
https://github.com/project-chip/connectedhomeip/issues/22038

Trying to build using : https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/android_building.md#preparing-for-build

1) Check out the Matter repository- **DONE**
2) Run bootstrap (only required first time) 
  - Not Working **source scripts/bootstrap.sh**
  - Facing This Issue :   https://github.com/project-chip/connectedhomeip/issues/22038

mohitjadav86, 2022-09-01:
this issue is done after successful build with **source scripts/bootstrap.sh**

closed 2022-09-01 by mohitjadav86