project-chip/connectedhomeip #22230: ACT-1.1 - Accepted commands lists only 1 entry value instead 12 entry

sowmyassp, 2022-08-29:
**Issue** : Accepted commands lists only 1 entry value instead 12 entry

**CommitID** - https://github.com/project-chip/connectedhomeip/commit/fa8db560907713c555b06120658034e8c6ccea1a (SVE-2 branch)

**Expected behaviour** : Accepted command list should 12 entries

**Actual behaviour** : Accepted command lists only 1 entry

**App used** - bridge-app

**Steps to reproduce** :
DUT side :
sudo rm -rf /tmp/chip_*
sudo ./chip-all-clusters-app

TH side
sudo rm -rf /tmp/chip_*
./chip-tool pairing ethernet 1 20202021 3840 fe80::e65f:1ff:fe0f:2753 5540
./chip-tool actions read accepted-command-list 1 1

**PFA Logs:**
[ACT.txt](https://github.com/project-chip/connectedhomeip/files/9444622/ACT.txt)

bzbarsky-apple, 2022-08-29:
@sowmyassp All the commands in this cluster are optional, so AcceptedCommandList will only contain them if they are actually supported, no?

tcarmelveilleux, 2022-08-29:
This is a bug with test script.

```
|6||| Read AcceptedCommandList (global attribute 65529) ; retain this list for reference in test [TC-{clusterid}-2.2] step 4d | list SHALL contain zero or more elements from 0..11 (and no other elements)

```

closed 2022-08-31 by cjandhyala