project-chip/connectedhomeip #22233: [ota] OTA requestor does not try another OTA provider when the current is down

Damian-Nordic, 2022-08-29:
#### Problem
https://github.com/project-chip/connectedhomeip/pull/16153/files has added the logic to re-try `QueryImage` in the case of timeout (which may be due to a lost CASE session), but there's no code to try another OTA Provider if the currently selected one is down or cannot be discovered at all 0. That is, OTA requestor will try another OTA Provider only after the query period (for example, 24 hours).

The behavior seems to be allowed by the spec (see below), but I wonder if it's intuitive and user-friendly:
> Handling errors from QueryImage Command
> (...)
> If the OTA Requestor is still attempting to discover an OTA Update, it MAY choose to attempt a QueryImage command on a different OTA Provider in its OTA Provider List

 #### Proposed Solution
Pick another OTA provider on connection failure.

Damian-Nordic, 2022-08-29:
@tcarmelveilleux @carol-apple @selissia I'm not sure what's severity of this. IMHO it's not critical, but I would love to hear your opinions.

doublemis1, 2022-08-29:
This issue impacts the SU-2.8 Certification test.
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/softwareupdate.adoc#361-tc-su-28-ota-functionality-in-multi-fabric-scenario

tcarmelveilleux, 2022-08-30:
@Damian-Nordic I'd say this issue is a matter of choice, but if integrators "blindly" use the SDK version, without doing sufficient failure trials, they may not realize that they have a less effective strategy

Damian-Nordic, 2022-08-30:
Thanks, test plan issue created: https://github.com/CHIP-Specifications/chip-test-plans/issues/2148, so I'll mark this as post 1.0.