project-chip/connectedhomeip #22251: can not pass the [TC-DD-2.1] in tv-app

nicelyjust, 2022-08-30:
#### Problem
when tested [TC-DD-2.1] Announcement by Device Verification [DUT - Commissionee], failed in step #14.b &14c , when execute the below command in chip-tool ,
```
avahi-browse _services._dns-sd._udp -r
```
can not get the Expected Outcome
```
Example output using all-clusters-app's advertisements:
11:30:36.040  Add        3   7 .                    _sub.local.          _L3840
11:30:36.040  Add        3   7 .                    _sub.local.          _S15
11:30:36.040  Add        2   7 .                    _sub.local.          _CM
```
 #### Proposed Solution

I'm not sure whether is a problem of the DNS-sd implement in Android platform, Appreciated for any response.

andy31415, 2022-09-06:
@nicelyjust Please update the bug with full description of steps and requirements. Specifically what are `#14b &14c` ? What commands are to be executed before that.

Please provide a list of commands for me to execute to reproduce the issue.
Also: what sha of the apps is being used?

andy31415, 2022-09-06:
The test seems to use RFC 6763 `9. Service Type Enumeration` (https://www.rfc-editor.org/rfc/rfc6763.html#page-27) to try to figure out if the device advertises the specified subtypes.

andy31415, 2022-09-06:
How can I run the tv-ap or tv-casting app to verify what is going on?

nicelyjust, 2022-09-07:
> 
Hi Andy:
precondition： ensure the tv-app or tv-casting app is commissioning state and is transport-connected ( Wi-Fi or Ethernet)
then execute the below command  to discover in your linux | macOS
```
avahi-browse _services._dns-sd._udp -r 
#or
dns-sd -B _services._dns-sd._udp -r
```

nicelyjust, 2022-09-07:
> The test seems to use RFC 6763 `9. Service Type Enumeration` (https://www.rfc-editor.org/rfc/rfc6763.html#page-27) to try to figure out if the device advertises the specified subtypes.

yes, I think so

andy31415, 2022-09-08:
I was told https://github.com/project-chip/connectedhomeip/blob/master/examples/tv-casting-app/android/README.md exists ... so that seems a good start, will try it.

andy31415, 2022-09-09:
Looking to debug this with `./out/linux-x64-minmdns/minimal-mdns-client`.

I observe results being returned on a query, but only on the first request. After that, it seems the query is not answered anymore.

[image]

andy31415, 2022-09-09:
Dnssd logic seems to have been added in #9884 by @xylophone21 . 

It uses NsdManger to register services. I am not sure if the behavior we are seeing is a regression or it never fully advertised. I believe it does advertise at boot, however I do not see NsdManager responding to queries that are done once the application runs.

andy31415, 2022-09-09:
It seems that service publishing does not keep a MulticastLock (but resolution does), so publishing is not persistent.

After fixing that, publish works however browsing for services will NOT return `_sub`. However running things like `./out/linux-x64-minmdns/minimal-mdns-client -q _T35._sub._matterc._udp.local` does yield results, so the data seems to be there.

andy31415, 2022-09-09:
I see:

```
> # This is a valid subtype
> avahi-browse _T35._sub._matterc._udp
+ wlp57s0 IPv6 2D1858D8557DB2FE                              _matterc._udp        local
+ wlp57s0 IPv4 2D1858D8557DB2FE                              _matterc._udp        local
+ enp58s0 IPv6 2D1858D8557DB2FE                              _matterc._udp        local
+ enp58s0 IPv4 2D1858D8557DB2FE                              _matterc._udp        local

> # This is NOT a valid subtype (no responses returned)
> avahi-browse _T36._sub._matterc._udp
^CGot SIGINT, quitting.
```

andy31415, 2022-09-09:
Regarding the test, https://www.rfc-editor.org/rfc/rfc6763.html#section-9 does not seem to say that subtypes are to be enumerated (but does not prevent it either).

I believe the test case needs updating or accepting alternate ways of verifying (like resolving the direct subtype like I did above).

lisaaaas, 2023-04-11:
I also encountered the same problem, running the instruction "avahi-browse _services._dns-sd. _udp-r" failed, what is the specific solution?

chrisdecenzo, 2023-05-24:
the fix improves the Android mDNS behavior but it does not address the issue with TC-DD-2.1 and TC-SC-4.1 since Android still does not respond to subtype queries.

The spec does not state that subtype queries are mandatory so probably the solution is to update TC-DD-2.1 and TC-SC-4.1 since these are currently mandatory.

tcarmelveilleux, 2023-10-11:
The fact that NsdManager cannot properly respond to the subtype query in some situations does not mean we need to remove/update the spec. It is possible on Android to be directly responding to particular queries, without going through NsdManager. Furthermore, NsdManager was improved in API level 33 and 34, and I would recommend that we separate platform support from spec requirements.

PR #22519 (2022-09-10): Get multicast lock when publishing services on android
#### Issue Being Resolved
* Fixes #22251

#### Change overview
Android 'publish' does not aquire a multicast lock, effectively disallowing the services from being seen on the network past any temporary "resolve" calls (which do aquire the lock).

Added aquire/release for the multicast lock when registering listeners.

1.0 justification: platform fix.