project-chip/connectedhomeip #22254: Build of docker images v0.5.98 from commit #22209 hung up

s07641069, 2022-08-30:
#### Problem
Build of docker images v0.5.98 from commit [#22209](https://github.com/project-chip/connectedhomeip/commit/5c2334a0d4ab173347cac2caad155fbb2f32d933) hung up

 #### Proposed Solution
Trigger build

s07641069, 2022-08-30:
Blocks https://github.com/project-chip/connectedhomeip/pull/22209

andy31415, 2022-08-30:
I clicked merge with master, to trigger new builds. This is Github infra having trouble and the only workaround I saw so far was to trigger more commits

s07641069, 2022-08-30:
@andy31415  Problem not with build of new pull request, it just blocked because of absence docker tag 0.5.98 which wasn't builded by next commit https://github.com/project-chip/connectedhomeip/commit/5c2334a0d4ab173347cac2caad155fbb2f32d933 for any vendor.
If you check build of docker images you will see that it in progress (hung up) almost 5 days.
[image]

s07641069, 2022-08-30:
Maybe we can increase version number here "integrations/docker/images/chip-build/version" without any other changes, just to trigger again docker images build?

andy31415, 2022-08-30:
The docker images are manually built. I started a 0.5.98 build now.

closed 2022-08-30 by s07641069