project-chip/connectedhomeip #22255: [SVE2] Chunking of attribute reports does not work with the ACL cluster

sstruchtrup, 2022-08-30:
#### Problem

Discovered when running a wildcard read on all attributes via `chip-tool any read-by-id 0xffffffff 0xffffffff 1 0xffff`.

If the ACL cluster tries to encode its attributes and cannot complete, due to the buffer space being exhausted, it will translate its internal error code `CHIP_ERROR_NO_MEMORY` into `CHIP_IM_GLOBAL_STATUS(ResourceExhausted)`. This is done in `ChipErrorToImErrorMap()`.

The report engine, namely the function `Engine::BuildSingleReportDataAttributeReportIBs()` will only test for CHIP_ERROR_BUFFER_TOO_SMALL and CHIP_ERROR_NO_MEMORY to enable chunking.

```c++
    if ((err == CHIP_ERROR_BUFFER_TOO_SMALL) || (err == CHIP_ERROR_NO_MEMORY))
    {
        ChipLogDetail(DataManagement, "<RE:Run> We cannot put more chunks into this report. Enable chunking.");

        //
        // Reset the error tracked within the builder. Otherwise, any further attempts to write
        // data through the builder will be blocked by that error.
        //
        attributeReportIBs.ResetError();
        err = CHIP_NO_ERROR;
    }
```
If an `CHIP_IM_GLOBAL_STATUS(ResourceExhausted)` is encountered, processing will fail and the error code will be propagated back over the wire to the requestor.

DUT log:
```
2022-08-30 12:55:38.139176 I DMG          Re-mapped src/system/TLVPacketBufferBackingStore.cpp:88: CHIP Error 0x0000000B: No memory into src/system/TLVPacketBufferBackingStore.cpp:88: CHIP Error 0x0000000B: No memory for IM return codes
2022-08-30 12:55:38.139204 I DMG          Error retrieving data from clusterId: 0x0000_001F, err = src/app/clusters/access-control-server/access-control-server.cpp:450: IM Error 0x00000589: General error: 0x89 (RESOURCE_EXHAUSTED)
2022-08-30 12:55:38.139236 I DMG          <RE:Run> Cluster 1f, Attribute fffc is dirty
```

chip-tool output:
```
# chip-tool any read-by-id 0xffffffff 0xffffffff 1 0xffff
...
[1661856926.330258][39297:39298] CHIP:DMG:              AttributeReportIB =
[1661856926.330264][39297:39298] CHIP:DMG:              {
[1661856926.330268][39297:39298] CHIP:DMG:                      AttributeStatusIB =
[1661856926.330273][39297:39298] CHIP:DMG:                      {
[1661856926.330278][39297:39298] CHIP:DMG:                              AttributePathIB =
[1661856926.330353][39297:39298] CHIP:DMG:                              {
[1661856926.330359][39297:39298] CHIP:DMG:                                      Endpoint = 0x0,
[1661856926.330367][39297:39298] CHIP:DMG:                                      Cluster = 0x1f,
[1661856926.330372][39297:39298] CHIP:DMG:                                      Attribute = 0x0000_0004,
[1661856926.330412][39297:39298] CHIP:DMG:                              }
[1661856926.330422][39297:39298] CHIP:DMG: 
[1661856926.330427][39297:39298] CHIP:DMG:                              StatusIB =
[1661856926.330468][39297:39298] CHIP:DMG:                              {
[1661856926.330475][39297:39298] CHIP:DMG:                                      status = 0x89 (RESOURCE_EXHAUSTED),
[1661856926.330480][39297:39298] CHIP:DMG:                              },
[1661856926.330518][39297:39298] CHIP:DMG: 
[1661856926.330524][39297:39298] CHIP:DMG:                      },
[1661856926.330530][39297:39298] CHIP:DMG: 
[1661856926.330534][39297:39298] CHIP:DMG:              },
[1661856926.330575][39297:39298] CHIP:DMG: 
...
[1661856938.197042][39301:39301] CHIP:TOO: Run command failure: IM Error 0x00000589: General error: 0x89 (RESOURCE_EXHAUSTED)
```

So far, we could unfortunately only reproduce this situation with our DUT, not with any of the example applications.

 #### Proposed Solution
Either have `Engine::BuildSingleReportDataAttributeReportIBs()` test for `CHIP_IM_GLOBAL_STATUS(ResourceExhausted)` or have the ACL cluster return `CHIP_ERROR_NO_MEMORY` or `CHIP_ERROR_BUFFER_TOO_SMALL`.

We fixed this with the following patch:

```diff
diff --git a/src/app/reporting/Engine.cpp b/src/app/reporting/Engine.cpp
index 792631f..b2f18da 100644
--- a/src/app/reporting/Engine.cpp
+++ b/src/app/reporting/Engine.cpp
@@ -189,7 +189,8 @@ CHIP_ERROR Engine::BuildSingleReportDataAttributeReportIBs(ReportDataMessage::Bu
                 // Otherwise, if partial data allowed, save the encode state.
                 // Otherwise roll back. If we have already encoded some chunks, we are done; otherwise encode status.

-                if (encodeState.AllowPartialData() && ((err == CHIP_ERROR_BUFFER_TOO_SMALL) || (err == CHIP_ERROR_NO_MEMORY)))
+                if (encodeState.AllowPartialData() && ((err == CHIP_ERROR_BUFFER_TOO_SMALL) || (err == CHIP_ERROR_NO_MEMORY) ||
+                        (err == CHIP_IM_GLOBAL_STATUS(ResourceExhausted))))
                 {
                     // Encoding is aborted but partial data is allowed, then we don't rollback and save the state for next chunk.
                     apReadHandler->SetAttributeEncodeState(encodeState);
@@ -201,7 +202,8 @@ CHIP_ERROR Engine::BuildSingleReportDataAttributeReportIBs(ReportDataMessage::Bu
                     attributeReportIBs.Rollback(attributeBackup);
                     apReadHandler->SetAttributeEncodeState(AttributeValueEncoder::AttributeEncodeState());

-                    if (err != CHIP_ERROR_NO_MEMORY && err != CHIP_ERROR_BUFFER_TOO_SMALL)
+                    if (err != CHIP_ERROR_NO_MEMORY && err != CHIP_ERROR_BUFFER_TOO_SMALL &&
+                        err != CHIP_IM_GLOBAL_STATUS(ResourceExhausted))
                     {
                         // Try to encode our error as a status response.
                         err = attributeReportIBs.EncodeAttributeStatus(pathForRetrieval, StatusIB(err));
@@ -241,7 +243,8 @@ exit:
     // These are are guaranteed to not fail since we've already reserved memory for the remaining 'close out' TLV operations in this
     // function and its callers.
     //
-    if ((err == CHIP_ERROR_BUFFER_TOO_SMALL) || (err == CHIP_ERROR_NO_MEMORY))
+    if ((err == CHIP_ERROR_BUFFER_TOO_SMALL) || (err == CHIP_ERROR_NO_MEMORY) ||
+            (err == CHIP_IM_GLOBAL_STATUS(ResourceExhausted)))
     {
         ChipLogDetail(DataManagement, "<RE:Run> We cannot put more chunks into this report. Enable chunking.");
```

bzbarsky-apple, 2022-08-30:
The right thing to fix here is the ACL cluster: ResourceExhausted could happen for reasons other than "no space in the ougoing buffer"...

@yunhanw-google

andy31415, 2022-08-30:
To me this sounds like a 1.0 blocker, but would like to have more oppinions. For now marked as patch acceptable, but would upgrade to blocker after more review.

bzbarsky-apple, 2022-08-30:
Steps to reproduce with all-clusters-app on Linux/Mac:

1. Set `CHIP_CONFIG_EXAMPLE_ACCESS_CONTROL_MAX_ENTRIES_PER_FABRIC` in `src/lib/core/CHIPConfig.h`
2. Run:
    ```
    chip-tool accesscontrol write acl '[{"privilege": 5, "authMode": 2, "subjects": [112233, 8589934592, 8589934593, 8589934594], "targets": [{"endpoint": 0, "cluster": null, "deviceType": null}, {"endpoint": 65534, "cluster": 131070, "deviceType": null}, {"endpoint": 65534, "cluster": 131070, "deviceType": null}]},{"privilege": 5, "authMode": 2, "subjects": [8589934591, 8589934592, 8589934593, 8589934594], "targets": [{"endpoint": 65534, "cluster": 131070, "deviceType": null}, {"endpoint": 65534, "cluster": 131070, "deviceType": null}, {"endpoint": 65534, "cluster": 131070, "deviceType": null}]},{"privilege": 5, "authMode": 2, "subjects": [8589934591, 8589934592, 8589934593, 8589934594], "targets": [{"endpoint": 65534, "cluster": 131070, "deviceType": null}, {"endpoint": 65534, "cluster": 131070, "deviceType": null}, {"endpoint": 65534, "cluster": 131070, "deviceType": null}]},{"privilege": 5, "authMode": 2, "subjects": [8589934591, 8589934592, 8589934593, 8589934594], "targets": [{"endpoint": 65534, "cluster": 131070, "deviceType": null}, {"endpoint": 65534, "cluster": 131070, "deviceType": null}, {"endpoint": 65534, "cluster": 131070, "deviceType": null}]},{"privilege": 5, "authMode": 2, "subjects": [8589934591, 8589934592, 8589934593, 8589934594], "targets": [{"endpoint": 65534, "cluster": 131070, "deviceType": null}, {"endpoint": 65534, "cluster": 131070, "deviceType": null}, {"endpoint": 65534, "cluster": 131070, "deviceType": null}]},{"privilege": 5, "authMode": 2, "subjects": [8589934591, 8589934592, 8589934593, 8589934594], "targets": [{"endpoint": 65534, "cluster": 131070, "deviceType": null}, {"endpoint": 65534, "cluster": 131070, "deviceType": null}, {"endpoint": 65534, "cluster": 131070, "deviceType": null}]},{"privilege": 5, "authMode": 2, "subjects": [8589934591, 8589934592, 8589934593, 8589934594], "targets": [{"endpoint": 65534, "cluster": 131070, "deviceType": null}, {"endpoint": 65534, "cluster": 131070, "deviceType": null}, {"endpoint": 65534, "cluster": 131070, "deviceType": null}]},{"privilege": 5, "authMode": 2, "subjects": [8589934591, 8589934592, 8589934593, 8589934594], "targets": [{"endpoint": 65534, "cluster": 131070, "deviceType": null}, {"endpoint": 65534, "cluster": 131070, "deviceType": null}, {"endpoint": 65534, "cluster": 131070, "deviceType": null}]},{"privilege": 5, "authMode": 2, "subjects": [8589934591, 8589934592, 8589934593, 8589934594], "targets": [{"endpoint": 65534, "cluster": 131070, "deviceType": null}, {"endpoint": 65534, "cluster": 131070, "deviceType": null}, {"endpoint": 65534, "cluster": 131070, "deviceType": null}]},{"privilege": 5, "authMode": 2, "subjects": [8589934591, 8589934592, 8589934593, 8589934594], "targets": [{"endpoint": 65534, "cluster": 131070, "deviceType": null}, {"endpoint": 65534, "cluster": 131070, "deviceType": null}, {"endpoint": 65534, "cluster": 131070, "deviceType": null}]},{"privilege": 5, "authMode": 2, "subjects": [8589934591, 8589934592, 8589934593, 8589934594], "targets": [{"endpoint": 65534, "cluster": 131070, "deviceType": null}, {"endpoint": 65534, "cluster": 131070, "deviceType": null}, {"endpoint": 65534, "cluster": 131070, "deviceType": null}]}]' ${NODE_ID} 0
    ```
3. Run `chip-tool accesscontrol read acl ${NODE_ID} 0`

andy31415, 2022-08-31:
Removed flags except 'patches acceptable' We have a PR to fix this so this should be good enough to continue.

closed 2022-08-31 by andy31415