project-chip/connectedhomeip #22257: Unable to checkout  in submodule path 'third_party/silabs/gecko_sdk'

mohitjadav86, 2022-08-30:
Used this Commands
- git clone https://github.com/project-chip/connectedhomeip.git
-  cd connectedhomeip 
- source scripts/bootstrap.sh
//Error while Clone
Unable to checkout  in submodule path 'third_party/silabs/gecko_sdk'

bzbarsky-apple, 2022-08-30:
@mohitjadav86 Did you `git submodule sync --recursive`?

mohitjadav86, 2022-08-31:
> @mohitjadav86 Did you `git submodule sync --recursive`?

Working now but getting someting like this

[BOOTSTRAP! Bootstrap may take a few minutes; please be patient.

Downloading and installing packages into local source directory:

  Setting up CIPD package manager...done (5.9s)
  Setting up Python environment.....done (6.7s)
  Setting up pw packages............skipped (0.1s)
  Setting up Host tools.............done (0.1s)

Activating environment (setting environment variables):

  Setting environment variables for CIPD package manager...done
  Setting environment variables for Python environment.....done
  **Setting environment variables for pw packages............skipped**
  Setting environment variables for Host tools.............done

Checking the environment:

20220831 10:54:46 INF Environment passes all checks!

Environment looks good, you are ready to go!

To reactivate this environment in the future, run this in your 
terminal:

  source ./activate.sh

To deactivate this environment, run this:

]

  **Setting environment variables for pw packages............skipped**  is it Okay ??

bzbarsky-apple, 2022-08-31:
> Setting environment variables for pw packages............skipped is it Okay ??

No idea, but I expect it is.  If you can compile, then it's ok.

mohitjadav86, 2022-08-31:
Okay, then issue is done. closing this issue.

closed 2022-08-31 by mohitjadav86