project-chip/connectedhomeip #22258: [SVE2][TC-SU-2.7] chip-tool interactive seems faulty

marius-alex-tache, 2022-08-30:
#### Problem
When subscribing to `state-transition` event in chip-tool interactive mode, I can only see the first two state transitions. After the new image is applied, if I restart the interactive shell and subscribe again, the third transition is also seen. 

I've also tried to subscribe with additional parameters: 
`otasoftwareupdaterequestor subscribe-event state-transition 5 10 2 0 --is-urgent true --keepSubscriptions true`. 

Although I can see the third state transition without restarting the shell, the fourth one can only be seen after another shell restart and subscribe. 
This does not seem normal, so please let me know if you have any insight regarding this matter.

bzbarsky-apple, 2022-08-30:
@marius-alex-tache what are the exact commands being run that show this problem?

marius-alex-tache, 2022-08-30:
Preconditions:
- DUT commissioned with node id 2
- OTA provider commissioned with node id 1

Verification plan steps:
```
./chip-tool interactive start
otasoftwareupdaterequestor subscribe-event state-transition 5 10 2 0
otasoftwareupdaterequestor announce-ota-provider 1 0 0 0 2 0
```
Expected outputs:
1. After OTA process starts. we should see the following transition events: 1 -> 2, 2 -> 4
2. After OTA finishes and the new image is applied, we should see the following transitions events: 4 -> 5, 5 -> 1
**1** happens, but no more state transition events are received after. 
If I now close this shell and start a new one:
```
./chip-tool interactive start
otasoftwareupdaterequestor subscribe-event state-transition 5 10 2 0
```
I can also see the other transitions, from **2**.

If I subscribe with `otasoftwareupdaterequestor subscribe-event state-transition 5 10 2 0 --is-urgent true --keepSubscriptions true` instead, seems like the shell is more robust and transition 4 -> 5 is also received after the new image is applied.

somu1710, 2022-08-30:
Hi @marius-alex-tache  Here state in events mentions the process of image while OTA update and  it will shows  which state it has been in currently and previously.

```
- In 1st event it shows previous state as 1 it means [Idle] and new state as 2 [querying] . 
- In 2nd event it shows previous state as 2 it means [Querying] and new state as 4 [downloading] .
- In 3rd event it shows previous state as 4 it means [downloading] and new state as 5 [applying] .
```

Once after the software update command we can able to verify the image has been querying and downloading.once after the image has been downloaded it will applying the image you can see this applying event after the image has been downloaded.In this 3 event we can verify the expected outcome of  state changes as [Idle,Querying,downloading and applying].

```
1.**Idle** - Its not in the process of update and its waiting for the query.
2.**Querying** - Its in the process of querying OTA provider
3.**Downloading** - Its in the process of downloading a softwareupdate
4.**Applying** - Its in process of verifying and applying software update
```

You mentioned that 4th event you get after the restart of the shell.Yes if OTA update has been finished after that event will show its in Idle position [New state- 1] as I mentioned above and [previous state -5] its in Applying state.Its correct and its working fine as expected.

Here I attach the spec link for your reference
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/service_device_management/OTASoftwareUpdate.adoc#7-ota-software-update-requestor-cluster-definition

bzbarsky-apple, 2022-08-31:
> otasoftwareupdaterequestor subscribe-event state-transition 5 10 2 0

OK, so that says "don't send me events more often than once every 5 seconds", right?  And in fact since it's non-urgent it will only send them every 10 seconds.

> After OTA finishes and the new image is applied,

This drops the subscription, since the server is restarted.  So you will get events that happened before the restart, modulo that 5 second hold-off which might prevent you from getting some of them if the restart happens too soon after the last-sent event.  You will get nothing after the restart.

>  If I now close this shell and start a new one:

Then you create a new subscription.

You could also wait for the existing subscription to discover it's timed out and create a new one without restarting chip-tool, to similar effect.

> If I subscribe with otasoftwareupdaterequestor subscribe-event state-transition 5 10 2 0 --is-urgent true --keepSubscriptions true

This should end up sending every 5 seconds, which yes, might get you more events close to shutdown.

marius-alex-tache, 2022-08-31:
@somu1710 @bzbarsky-apple Thanks for the clarifications!

closed 2022-08-31 by marius-alex-tache