project-chip/connectedhomeip #22260: Cannot find source file

uiabhishek15, 2022-08-30:
#### Problem
<C:\Users\abhis\Downloads\sdk-connectedhomeip-2.0.2\sdk-connectedhomeip-2.0.2\out\android_arm64\CMakeLists.txt : C/C++ debug|arm64-v8a : CMake Error at C:\Users\abhis\Downloads\sdk-connectedhomeip-2.0.2\sdk-connectedhomeip-2.0.2\out\android_arm64\CMakeLists.txt:19 (add_library):
  Cannot find source file:

    C:/Users/abhis/Downloads/sdk-connectedhomeip-2.0.2/sdk-connectedhomeip-2.0.2/src/android/CHIPTool/appsrc/main/cpp/native-lib.cpp

  Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
  .hxx .in .txx>

uiabhishek15, 2022-08-30:
Can anybody please help me with this?

bzbarsky-apple, 2022-08-30:
Looks a lot like https://github.com/project-chip/connectedhomeip/issues/22256 but for a different filename.

uiabhishek15, 2022-08-30:
I solved this issue but again I am getting the below error
ninja: error:
 'C:/Users/abhis/Downloads/sdk-connectedhomeip-2.0.2/setup_payload/libSetupPayload.a', 
needed by 
'C:/Users/abhis/Downloads/sdk-connectedhomeip-2.0.2/sdk-connectedhomeip2.0.2/src/android/CHIPTool/app/build/intermediates/cmake/debug/obj/arm64-v8a/libnative-lib.so',
 missing and no known rule to make it

@bzbarsky-apple  Can you please help me with this?

bzbarsky-apple, 2022-08-31:
@uiabhishek15 I have no idea about the Android bits, sorry...

mohitjadav86, 2022-08-31:
> I solved this issue but again I am getting the below error ninja: error: 'C:/Users/abhis/Downloads/sdk-connectedhomeip-2.0.2/setup_payload/libSetupPayload.a', needed by [...]

How you Solve this ???

uiabhishek15, 2022-09-01:
@mohitjadav86 I didn't solved the ninja error, if you solve this let me know and can anyone Android developers help me with this?

mohitjadav86, 2022-09-01:
**ninja error missing and no known rule to make it**
Happens in my case of java version try to downgrade java version

```
# List Java versions installed
/usr/libexec/java_home -V

# Java 11
export JAVA_HOME=$(/usr/libexec/java_home -v 11)

# Java 1.8
export JAVA_HOME=$(/usr/libexec/java_home -v 1.8)

# Java 1.7
export JAVA_HOME=$(/usr/libexec/java_home -v 1.7)

# Java 1.6
export JAVA_HOME=$(/usr/libexec/java_home -v 1.6)
```

use java 11

mohitjadav86, 2022-09-01:
> @mohitjadav86 I didn't solved the ninja error, if you solve this let me know and can anyone Android developers help me with this?

yes working now https://github.com/project-chip/connectedhomeip/issues/22260#issuecomment-1234219127

closed 2022-09-06 by andy31415