project-chip/connectedhomeip #22265: [TC-WNCV-3.1] CurrentPositionLiftPercent100ths attribute read value mismatch error

manjunath-grl, 2022-08-30:
Summary:
As per testplan of WNCV-2.1 in Step 3b: TH reads CurrentPositionLiftPercent100ths  and the expected outcome value must not be 10000 ( 0x2710 ) but got 10000 when executed.
Steps needs review 3b , 3c, 3d, 3e

Diff link: https://github.com/project-chip/connectedhomeip/compare/master...manjunath-grl:connectedhomeip:CurrentPositionLiftPercent100ths_Value_Mismatch

Testplan screenshot:
[image]

Execution logs:
[WNCV_3_1.txt](https://github.com/project-chip/connectedhomeip/files/9454170/WNCV_3_1.txt)

bzbarsky-apple, 2022-08-30:
@manjunath-grl Where are the "TH waits for 1-5 seconds" bits in the test plan coming from?  The test plan is assuming that the movement will not complete in that time, but why is it assuming that?

bzbarsky-apple, 2022-08-30:
Ah, I guess this is moving toward 0, so the assertion is that it has moved at least some?

Is this failing reliably, or just some of the time @manjunath-grl ?

manjunath-grl, 2022-08-30:
@bzbarsky-apple this is failing every time when executed.

bzbarsky-apple, 2022-08-30:
So the log shows that we send the UpOrOpen command (at a point when LiftPercent100ths is 10000) then we do our reads, etc, and then after all that LiftPercent100ths is set to 0.  I don't see it being set to any intermediate values.

@jmeg-sfy any idea why that's happening?

jmeg-sfy, 2022-08-31:
There is only one single shot timer on all-Cluster-app to fake a motion update (w/o intermediate values)
#define FAKE_MOTION_DELAY_MS 5000  here src/app/clusters/window-covering-server/window-covering-server.cpp
it is defined to update to the final value after 5s . this is not the best way to simulate it for sure but it worked

@manjunath-grl  keep step "3a2: DUT updates its attributes" to 2000 ms or increase to 3000 ms

jmeg-sfy, 2022-08-31:
@manjunath-grl https://github.com/project-chip/connectedhomeip/pull/22284 i made some comments to solve this issue

bzbarsky-apple, 2022-08-31:
> There is only one single shot timer on all-Cluster-app to fake a motion update (w/o intermediate values)

It's not clear to me that that's a legal implementation of  the SHALL bits of "5.3.4.3. Position Aware" in appclusters...

jmeg-sfy, 2022-08-31:
Position aware just require to have at the end (current == target) position
but a real device will update its current along the way.
We can improve all-cluster-app for CI or better create a dedicated window-app later post v1.0

PR #22497 (2022-09-15): Fix CurrentPositionLiftPercent100ths attribute read value mismatch
* Currently, cert Test_TC_WNCV_3_1 and Test_TC_WNCV_3_2 are failed and temporarily disabled. This is because the current mock implementation of movement is one shot and up to the end straight after 5s, but the test requires the value change gradually when window is opened or closed.

* Current mock implementation of window covering movement is done in WindowCovering cluster Server, this is not right, since the cluster server is used in both example app and product. 

#### Issue Being Resolved
* Fixes #22265, #22287

#### Change [... cut]
closed 2022-09-15 by andy31415