project-chip/connectedhomeip #22270: We need to release the buffer in case of failure.

yufengwangca, 2022-08-30:
#### Problem
 
 We need to release the buffer in case of failure.

```
void UDPEndPointImplLwIP::LwIPReceiveUDPMessage(void * arg, struct udp_pcb * pcb, struct pbuf * p, const ip_addr_t * addr,
                                                u16_t port)
{

    CHIP_ERROR err = ep->GetSystemLayer().ScheduleLambda(
        [ep, p = System::LwIPPacketBufferView::UnsafeGetLwIPpbuf(buf), pktInfo = pktInfo.get()] {
            ep->HandleDataReceived(System::PacketBufferHandle::Adopt(p), pktInfo);
        });

    if (err == CHIP_NO_ERROR)
    {
        // If ScheduleLambda() succeeded, it has ownership of the buffer, so we need to release it (without freeing it).
        static_cast<void>(std::move(buf).UnsafeRelease());
        // Similarly, ScheduleLambda now has ownership of pktInfo.
        pktInfo.release();
    }
}
```

 #### Proposed Solution
<suggested fix, suggested enhancement>

bzbarsky-apple, 2022-08-30:
@yufengwangca In case of failure, won't the `PacketBufferHandle` destructor release the buffer?

andy31415, 2022-09-01:
I have a somewhat hard time understanding the issue: what buffer is this about? the current implementation has a `Platform::UniquePtr<struct pbuf> pbufFreeGuard(p);` so pbuf should be freed.

andy31415, 2022-09-01:
I am somewhat unclear about lifetimes of `buf` in general though:
  - `PacketBufferHandle` creates it and owns its lifetime
  - `UnsafeGetLwIPpbuf` has usage but does NOT own it (according to the documentation of that function)
  - lambda will access it

so on lambda schedule success, who owns the buffer? Will PacketBufferHandle free it and if so, why is the raw pointer inside the lambda still valid and available for adoption?

andy31415, 2022-09-01:
I am unclear which buffer we need releasing in case of failure - all seems ok to me.

yufengwangca, 2022-09-01:
Thanks, in case of failure, buf will be auto released by PacketBufferHandle,  pbuf * p and pktInfo are both handled by UniquePtr, it looks no leak in this case.

closed 2022-09-01 by yufengwangca