project-chip/connectedhomeip #22272: Seg fault in src/app/util/attribute-storage.cpp

latch-danielkneip, 2022-08-30:
#### Problem
We’ve run into a seg fault in src/app/util/attribute-storage.cpp when emAfEndpoints[ep].endpointType is 0 and emberAfFindClusterInType() is called from emberAfFindClusterIncludingDisabledEndpoints(). Adding  emAfEndpoints[ep].endpointType != nullptr before the call allows us to function normally. However, I’m concerned there may be a deeper issue.

The seg fault occurs after issuing the following command to dynamic endpoint 4 that represents a bridged dimmer:
```
./chip-tool onoff toggle 1 4
```

 #### Proposed Solution
```
diff --git a/src/app/util/attribute-storage.cpp b/src/app/util/attribute-storage.cpp
index 7e8bad312..7d3124d30 100644
--- a/src/app/util/attribute-storage.cpp
+++ b/src/app/util/attribute-storage.cpp
@@ -796,7 +796,7 @@ const EmberAfCluster * emberAfFindClusterIncludingDisabledEndpoints(EndpointId e
                                                                     EmberAfClusterMask mask)
 {
     uint16_t ep = emberAfIndexFromEndpointIncludingDisabledEndpoints(endpoint);
-    if (ep < MAX_ENDPOINT_COUNT)
+    if (ep < MAX_ENDPOINT_COUNT && emAfEndpoints[ep].endpointType != nullptr)
     {
         return emberAfFindClusterInType(emAfEndpoints[ep].endpointType, clusterId, mask);
     }
```

bzbarsky-apple, 2022-08-30:
If we don't fix this for 1.0, I expect we will have common crashes on any app that allocates some dynamic endpoints but does not actually have endpoints defined for all of them (e.g. any bridge that is not maxed out in terms of the number of devices it's bridging).

andy31415, 2022-08-30:
Master patch acceptable: segfault fix.

PR #22275 (2022-08-31): Fix endpoint index lookups to handle invalid endpoint ids correctly.
There are some code paths that loop over all endpoint indices (including ones
that do not have an endpoint defined yet) and then use the endpoint id to look
up endpoint indices.  Make sure we don't claim an endpoint index for the "not an
endpoint" endpoint id.

Also fixes findClusterEndpointIndex to check for the invalid endpoint id before
working with it, since that means the endpoint index it's looking at does not
correspond to a defined endpoint.

Fixes https://github.com/project-chip/connectedhomeip/issues/22272

#### Pro [... cut]
closed 2022-08-31 by bzbarsky-apple