project-chip/connectedhomeip #22287: [TC-WNCV-3.2] CurrentPositionLiftPercent100ths attribute read value mismatch error

manjunath-grl, 2022-08-30:
Summary:
As per testplan of WNCV-3.2 in Step 3b: TH reads CurrentPositionLiftPercent100ths  and the expected outcome value must not be 0 ( 0x0 ) but got 0 when executed.
Steps needs review 3b , 3c, 3d, 3e
Client app: All clusters app
Server app: Chip-tool

Diff link:https://github.com/project-chip/connectedhomeip/compare/master...manjunath-grl:connectedhomeip:WNCV_3_2_Value_Mismatch

Testplan screenshot:
[image]

Execution logs:
[WNCV_3_2.txt](https://github.com/project-chip/connectedhomeip/files/9455969/WNCV_3_2.txt)

bzbarsky-apple, 2022-08-30:
This sure looks like the same thing as https://github.com/project-chip/connectedhomeip/issues/22265 but going the other direction.  Is it possible the directions are reversed?

andy31415, 2022-08-31:
manjunath-grl - why is the test saying that a current position percentage cannot be 0%? 

I am missing test steps and description here. 0% seems like a valid percentage to me.

andy31415, 2022-08-31:
From the description of #22265 it looks like the mock implementation in all clusters has a single-shot timer for updating location.

Tests seem to expect some more gradual movement without hitting the endpoints immediatley, hence the error.

Marking it as "patches to master acceptable" (generally true since this probably only affects /examples folder) however not blocking 1.0 release.