project-chip/connectedhomeip #22288: Actions cluster - ChipTool throws (UNSUPPORTED_COMMAND) for [TC-ACT-3.2]

phonnakasturi-apple, 2022-08-31:
With SHA 5471b35f09beafd7db4a5e6ccb9ca2e63d7c9010

For [TC-ACT-3.2] Functionality with client as DUT, chip-tool throws 
`Run command failure: IM Error 0x00000581: General error: 0x81 (UNSUPPORTED_COMMAND)`

**Steps to reproduce**

1. Run ./chip-all-clusters-app  on raspi
2. Pari chip tool
3. Execute the command mentioned in the test case -> ./chip-tool actions start-action-with-duration 0x1001 100 1 1

**Attaching the logs**

[TC-ACT-3.2-THLogs.txt](https://github.com/project-chip/connectedhomeip/files/9457388/TC-ACT-3.2-THLogs.txt)
[TC-ACT-3.2-DUTLogs.txt](https://github.com/project-chip/connectedhomeip/files/9457394/TC-ACT-3.2-DUTLogs.txt)

bzbarsky-apple, 2022-08-31:
chip-all-clusters-app does not support start-action-with-duration, which is why you get UNSUPPORTED_COMMAND.

andy31415, 2022-08-31:
I do not believe we can support new clusters in time for 1.0.

phonnakasturi-apple, 2022-08-31:
Updated the app name it must be bridge-app (sorry about the typo earlier)

bzbarsky-apple, 2022-08-31:
bridge-app does not support start-action-with-duration either, fwiw.

cjandhyala, 2022-09-01:
closing the issue

closed 2022-09-01 by cjandhyala