project-chip/connectedhomeip #22290: [SVE2][TC-CC-6.3] Step-color-temperature does not reach ColorTemperaturePhysicalMaxMIreds

pankore, 2022-08-31:
#### Problem
- Reading ColorTemperaturePhysicalMaxMireds gives 65729.
- Step 2b, after running command `./chip-tool colorcontrol step-color-temperature 1 32639 200 0 65279 0 0 1 1` and waiting more than 25 seconds, reading the color-temperature gives 65278 instead of the ColorTemperaturePhysicalMaxMireds (65279).

andy31415, 2022-08-31:
@pankore - what is the DUT you are testing with? Can it be reproduced for developers in one of the standard apps (e.g. light on linux or all-clusters or linux or something like that)?

What is the SHA you are using to run this test?

mkardous-silabs, 2022-09-01:
@pankore  @andy31415 
I tried reproducing this issue but I do not think this is a bug. This devices behavior is correct.

In the test verification steps, TC-CC-6.3 2A is `./chip-tool colorcontrol move-to-color-temperature 32639 0 0 0 1 1`
This command moves the `color-temperature` to 32639 which represents a step from 0 to `floor(65279 / 2)`.

The next test step, 2B, is `./chip-tool colorcontrol step-color-temperature 1 32639 200 0 65279 0 0 1 1` which does a step of `32639` over `20s`. This represents a move of `floor(65279 / 2)` from `32639`.

After both steps, the device will have moved it's color temperature two times of `32639` which represents a total of `65278`.
The misunderstanding is coming from the fact that the device cannot move of `32639.5 = 65279 / 2` The test verification formula in step 2a, `ColorTemperatureMireds=(ColorTempPhysicalMinMireds + ColorTempPhysicalMaxMireds)/2`, is slightly misleading because of the rounding that happens when the sum of min / max is an odd number.

andy31415, 2022-09-01:
@cjandhyala - looks like a test script error from the description of @mkardous-silabs

mkardous-silabs, 2022-09-01:
Based on the description of the test : 
> StepColorTemperature (up) starting midway between min/max; use a step size that will take (about) 20s to get to max

The steps should be :
2a : `./chip-tool colorcontrol move-to-color-temperature 32639 0 0 0 1 1`
2b: `./chip-tool colorcontrol step-color-temperature 1 32640 200 0 65279 0 0 1 1`

so yes, this is a test plan / script issue.

mkardous-silabs, 2022-09-01:
I opened at test plan issue so this issue can be closed
https://github.com/CHIP-Specifications/chip-test-plans/issues/2175

sankarselvam, 2022-09-02:
@pankore  and @mkardous-silabs , I commented in the test plan ticket. This is not a SDK issue. 

Also VS command has been given as per the test plan. We can add additional clarification in the testplan to reduce any confusion which, I have already commented in the test plan issue https://github.com/CHIP-Specifications/chip-test-plans/issues/2175

mkardous-silabs, 2022-09-02:
@andy31415 This issue can be closed.

cjandhyala, 2022-09-02:
closing the issue based the comments

closed 2022-09-02 by cjandhyala

leorozendaal, 2023-02-15:
@raju-apple @manjunath-grl @yinyihu-silabs @ukatira @mkardous-silabs 
Two comments
- all these tests are governed by the "allow 15% tolerance (of the total step)" so a difference of 1 on a step of (about 32768) is perfectly fine. If the test script does not allow for a tolerance, it needs to be added. test plan does not need to be updated
- I am seriously wondering what DUT you are testing. No realistic product will have such a huge range of CT values.