project-chip/connectedhomeip #22302: [SVE2][TC-IDM-4.1] Test Plan and Test Script issue

jysalaun, 2022-08-31:
After following all the first manual commands in chip-tool interactive mode, the last one fails
<both commands in Excel file and in tc_idm_4.1.py (assuming mandatory "\" are removed):

any subscribe-by-id "6,8,3" "0,1,0" 10 100 "1" "1,1,1" from Excel
any subscribe-by-id \"6,8,3\" \"0,1,0\" 10 100 \"1\" \"1,1,1\" from tc_idm_4.1.py

raju-apple, 2022-08-31:
@cjandhyala do we need to change the command ?

bzbarsky-apple, 2022-08-31:
@jysalaun What fails?  Please attach logs.

cjandhyala, 2022-09-01:
I prefer to close the issue here in this repo and have @jysalaun open the issue in test plan/test scripts repo. Closing the issue here, @jysalaun Reopen the issue if you think there is a SDK bug.

closed 2022-09-01 by cjandhyala

bzbarsky-apple, 2022-09-01:
Ah, per question earlier today on slack, this is being run in interactive mode.

In interactive mode, the quotes used for arguments are single quotes (so then you can use `"` in your JSON for complex args); it's not the shell.  So the right command would be either:
```
any subscribe-by-id '6,8,3' '0,1,0' 10 100 1 '1,1,1'
```
or just
```
any subscribe-by-id 6,8,3 0,1,0 10 100 1 1,1,1
```