project-chip/connectedhomeip #22308: OTA on linux does not support BlockEOF of data length zero.

rgoliver, 2022-08-31:
#### Problem
According to the spec:

> [11.21.6.5. BlockEOF Message]
> The BlockEOF message represents the final block in a data transfer.
> 
> Note that, unlike a Block, BlockEOF MAY have a data length of zero. If the entire transfer fits within the negotiated block size, the BlockEOF SHALL be the one and only message sent in the exchange and no Block messages will be sent. In that trivial case, the Block Counter would be 0 in the BlockEOF.

Linux implementation currently doesn't allow this:
 
[src/platform/Linux/OTAImageProcessorImpl.cpp, line 71](https://github.com/project-chip/connectedhomeip/blob/master/src/platform/Linux/OTAImageProcessorImpl.cpp#L71)
```
    if ((block.data() == nullptr) || block.empty())
    {
        return CHIP_ERROR_INVALID_ARGUMENT;
    }
```

 #### Proposed Solution
Remove this check, it is checked immediately after in SetBlock: https://github.com/project-chip/connectedhomeip/blob/master/src/platform/Linux/OTAImageProcessorImpl.cpp#L244

But at this point the block is released correctly and it does not return an error: 
```
    if (!IsSpanUsable(block))
    {
        ReleaseBlock();
        return CHIP_NO_ERROR;
    }
 ```
 
 This also mimics how other  platforms are implemented.

andy31415, 2022-08-31:
Marking accepting patches for platform-specific fixes for OTA.
This is borderline 'blocking v1' I think, however since we have a PR already in this should be sufficient.

franck-apple, 2022-09-01:
Closing, as #22314 merged.

closed 2022-09-01 by andy31415