project-chip/connectedhomeip #22311: [SVE2][TC-ACT-3.1] Test Plan and Test Script issue

jysalaun, 2022-08-31:
#### Problem
<wrong chip-tool commands in both tc_act_3.1.py and Excel sheet:
  * _chip-tool actions  read endpoint-list  1 1_  raises an error from chip-tool
 * _chip-tool bridgedactions  read action-list 1 1_  raises an error from chip-tool
 * _chip-tool bridgedactions  read endpoint-list   1 1_  raises an error from chip-tool

>

 #### Proposed Solution
<suggested fix, suggested enhancement>
 * _chip-tool actions  read endpoint-list  1 1_ should be _chip-tool actions  read **endpoint-lists**   1 1_
and so on

bzbarsky-apple, 2022-08-31:
@jysalaun Please file test plan and test script issues in https://github.com/CHIP-Specifications/chip-test-plans/issues

aswathygrl, 2022-09-30:
@jysalaun @raju-apple Verification sheet has already been Updated , could you please close this issue?

closed 2022-10-04 by jysalaun