project-chip/connectedhomeip #22330: [TC-CADMIN-1.8] unable to run the command ./chip-tool administratorcommissioning open-basic-commissioning-window 500 2 0 --timedInteractionTimeoutMs 1000

satish-hsk, 2022-09-01:
#### Problem : Unable to run the ./chip-tool administratorcommissioning open-basic-commissioning-window 500 2 0 --timedInteractionTimeoutMs 1000 
on our platform
 * expected behavior: A response when we run this command
 * actual behavior: No response from RPi
 * steps to reproduce, Commission and then run the above command
 * system configuration
[Software component]
[Raspi] Running chip-all-cluster-app
[Mobile (our DUT as Commissioner)]: After successful commissioning, issued the above command 
 * screenshots, images, logs, etc.

satish-hsk, 2022-09-01:
bonito:/data/chip-tool-bionic # ./chip-tool administratorcommissioning open-basic-commissioning-window 500 2 0 --timedInteractionTimeoutMs 1000 
CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /data/local/tmp/chip_kvs
CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /data/local/tmp/chip_factory.ini
CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /data/local/tmp/chip_config.ini
CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /data/local/tmp/chip_counters.ini
CHIP:DL: writing settings to file (/data/local/tmp/chip_counters.ini-VwhLL5)
CHIP:DL: renamed tmp file to file (/data/local/tmp/chip_counters.ini)
CHIP:DL: NVS set: chip-counters/reboot-count = 615 (0x267)
CHIP:DL: Failed to get Ethernet interface
CHIP:DL: Failed to reset Ethernet statistic counts
CHIP:DL: Got WiFi interface: wlan0
CHIP:DL: Failed to reset WiFi statistic counts
CHIP:IN: UDP::Init bind&listen port=0
CHIP:IN: UDP::Init bound to port=45417
CHIP:IN: UDP::Init bind&listen port=0
CHIP:IN: UDP::Init bound to port=39203
CHIP:IN: BLEBase::Init - setting/overriding transport
CHIP:IN: TransportMgr initialized
CHIP:FP: Initializing FabricTable from persistent storage
CHIP:TS: Last Known Good Time: 2022-08-23T16:38:14
CHIP:FP: Fabric index 0x1 was retrieved from storage. Compressed FabricId 0x5EFC34E6354DC315, FabricId 0x0000000000000001, NodeId 0x000000000001B669, VendorId 0xFFF1
CHIP:FP: Fabric index 0x2 was retrieved from storage. Compressed FabricId 0x928D2843C8C16D64, FabricId 0x0000000000000002, NodeId 0x000000000001B669, VendorId 0xFFF1
CHIP:ZCL: Using ZAP configuration...
CHIP:DL: MDNS failed to join multicast group on r_rmnet_data0 for address type IPv4: ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:758: Inet Error 0x00000110: Address not found
CHIP:DL: MDNS failed to join multicast group on rmnet_data0 for address type IPv4: ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:758: Inet Error 0x00000110: Address not found
CHIP:DL: MDNS failed to join multicast group on rmnet_ipa0 for address type IPv4: ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:758: Inet Error 0x00000110: Address not found
CHIP:DL: MDNS failed to join multicast group on dummy0 for address type IPv4: ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:758: Inet Error 0x00000110: Address not found
CHIP:CTL: System State Initialized...
CHIP:CTL: Stopping commissioning discovery over DNS-SD
CHIP:CTL: Setting attestation nonce to random value
CHIP:CTL: Setting CSR nonce to random value
CHIP:IN: UDP::Init bind&listen port=5550
CHIP:IN: UDP::Init bound to port=5550
CHIP:IN: UDP::Init bind&listen port=5550
CHIP:IN: UDP::Init bound to port=5550
CHIP:IN: TransportMgr initialized
CHIP:DL: CHIP task running
CHIP:CTL: Stopping commissioning discovery over DNS-SD
CHIP:CTL: Setting attestation nonce to random value
CHIP:CTL: Setting CSR nonce to random value
CHIP:CTL: Generating NOC
CHIP:FP: Validating NOC chain
CHIP:FP: NOC chain validation successful
CHIP:FP: Updated fabric at index: 0x1, Node ID: 0x000000000001B669
CHIP:TS: Last Known Good Time: 2022-08-23T16:38:14
CHIP:TS: New proposed Last Known Good Time: 2021-01-01T00:00:00
CHIP:TS: Retaining current Last Known Good Time
CHIP:FP: Metadata for Fabric 0x1 persisted to storage.
CHIP:TS: Committing Last Known Good Time to storage: 2022-08-23T16:38:14
CHIP:CTL: Joined the fabric at index 1. Compressed fabric ID is: 0x0000000000000000
CHIP:IN: UDP::Init bind&listen port=5550
CHIP:IN: UDP::Init bound to port=5550
CHIP:IN: UDP::Init bind&listen port=5550
CHIP:IN: UDP::Init bound to port=5550
CHIP:IN: TransportMgr initialized
CHIP:TOO: Sending command to node 0x2
CHIP:CSM: FindOrEstablishSession: PeerId = [1:0000000000000002]
CHIP:CSM: FindOrEstablishSession: No existing OperationalSessionSetup instance found
CHIP:CTL: OperationalSessionSetup[1:0000000000000002]: State change 1 --> 2
CHIP:DIS: Warning: Attempt to mDNS broadcast failed on rmnet_ipa0:  ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:406: OS Error 0x02000065: Network is unreachable
CHIP:DIS: mDNS broadcast had only partial success: 9 successes and 1 failures.
CHIP:DIS: Checking node lookup status after 206 ms
CHIP:DIS: Warning: Attempt to mDNS broadcast failed on rmnet_ipa0:  ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:406: OS Error 0x02000065: Network is unreachable
CHIP:DIS: mDNS broadcast had only partial success: 5 successes and 1 failures.
CHIP:DIS: Warning: Attempt to mDNS broadcast failed on rmnet_ipa0:  ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:406: OS Error 0x02000065: Network is unreachable
CHIP:DIS: mDNS broadcast had only partial success: 5 successes and 1 failures.
CHIP:DIS: Warning: Attempt to mDNS broadcast failed on rmnet_ipa0:  ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:406: OS Error 0x02000065: Network is unreachable
CHIP:DIS: mDNS broadcast had only partial success: 5 successes and 1 failures.
CHIP:CTL: Shutting down the commissioner
CHIP:CTL: Stopping commissioning discovery over DNS-SD
CHIP:CTL: Shutting down the controller
CHIP:DIS: OperationalSessionSetup[1:0000000000000002]: Cancelling incomplete address resolution as device is being deleted.
CHIP:IN: Expiring all sessions for fabric 0x1!!
[... cut ...]
CHIP:BLE: BleConnectionDelegate::CancelConnection is not implemented.
CHIP:DL: writing settings to file (/data/local/tmp/chip_counters.ini-OYN3TB)
CHIP:DL: renamed tmp file to file (/data/local/tmp/chip_counters.ini)
CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
CHIP:DL: Inet Layer shutdown
CHIP:DL: BLE shutdown
CHIP:DL: System Layer shutdown
CHIP:TOO: Run command failure: ../examples/chip-tool/commands/common/CHIPCommand.cpp:454: CHIP Error 0x00000032: Timeout
1|bonito:/data/chip-tool-bionic #

satish-hsk, 2022-09-01:
updated the nodeid to the correct value 1 and it made it work, please update the test case
 ./chip-tool administratorcommissioning open-basic-commissioning-window 500 **1** 0 --timedInteractionTimeoutMs 1000

andy31415, 2022-09-01:
Marking as test case issue.

@cjandhyala this seems like it needs a test case update.

andy31415, 2022-09-01:
test case fixes are acceptable for 1.0 fixes

franck-apple, 2022-09-09:
Per discussion with @cjandhyala, the test environment needed to be updated. Closing.

closed 2022-09-09 by franck-apple