project-chip/connectedhomeip #22333: Darwin subscribeAttribute APIs leak

bzbarsky-apple, 2022-09-01:
#### Problem
See https://github.com/project-chip/connectedhomeip/pull/22323#issuecomment-1233705089

Basically, using the ReadInteraction SubscribeAttribute API (via CHIPCluster in this case) means that there is no way to free the callback context (except on errors), because it never delivers OnDone.

In particular, this means that if you shut down and restart the stack the subscription will be dead in the water but its associated memory will be leaked.

This has been a problem all along, but it got highlighted in #22323 because we started freeing the ReadClient (but not the callback context) and now LSan can detect the leak.  I assume that it couldn't before because it can't detect leaks of objects that reference each other; it thinks they are referenced from allocated memory....

 #### Proposed Solution
Either fix the ReadInteraction and CHIPCluster API to allow passing in an OnDone callback for subscriptions or stop using them altogether.

Or remove the Darwin subscribeAttribute APIs.

bzbarsky-apple, 2022-09-01:
When this is fixed, we should remove the relevant suppression in `scripts/tests/chiptest/lsan-mac-suppressions.txt`.

jtung-apple, 2022-09-15:
Darwin: moving to v1.1

closed 2022-10-26 by jtung-apple