project-chip/connectedhomeip #22358: minimal mDNS receives SRV+AAAA+TXT records during operational parse, but mActiveLookups is empty

AlanLCollins, 2022-09-02:
#### Problem
during back to back pairing Thread accessory devices (e.g. smartplug / light-bulb), I get "CHIP DIS: Operational discovery failed" 
I turned on the min-mDNS CHIP_MINMDNS_HIGH_VERBOSITY = true + I injected my own debug logs, and I see the following behavior:
```
09-02 05:53:34.665  1061  1608 I CHIP API: aceChip_clusterSendCommand
09-02 05:53:34.665  1061  1608 I CHIP API: aceChip_clusterSendCommand() confirmation_id=26
09-02 05:53:34.665  1061  1608 I CHIP CSM: FindOrEstablishSession: PeerId = A4575F13C9B1C48B:024E7EEE8038E58B
09-02 05:53:34.665  1061  1608 I CHIP CSM: FindOrEstablishSession: No existing OperationalDeviceProxy instance found
09-02 05:53:34.665  1061  1608 I CHIP CTL: OperationalDeviceProxy[A4575F13C9B1C48B:024E7EEE8038E58B]: State change 1 --> 2
09-02 05:53:34.665  1061  1608 I CHIP DIS: MINMDNS: Sending query ANY/UNICAST for A4575F13C9B1C48B-024E7EEE8038E58B._matter._tcp.local
09-02 05:53:34.666  1061  3877 E CHIP DIS: Truncated responses not supported for address resolution
09-02 05:53:34.666  1061  3877 I CHIP DIS: MINMDNS: received SRV record for A4575F13C9B1C48B-024E7EEE8038E58B._matter._tcp.local
09-02 05:53:34.666  1061  3877 I CHIP DIS: MINMDNS:     Operational SRV for A4575F13C9B1C48B-024E7EEE8038E58B: 2A126A0633994574.local
09-02 05:53:34.666  1061  3877 E CHIP DIS: Truncated responses not supported for address resolution
09-02 05:53:34.666  1061  3877 I CHIP DIS: MINMDNS: received AAAA record for 2A126A0633994574.local
09-02 05:53:34.666  1061  3877 I CHIP DIS: MINMDNS:     IP address fdc5:5696:c095:1:ab6e:723f:3690:6cdd found for 2A126A0633994574.local
09-02 05:53:34.666  1061  3877 I CHIP DIS: MINMDNS: received TXT record for A4575F13C9B1C48B-024E7EEE8038E58B._matter._tcp.local
09-02 05:53:34.666  1061  3877 E CHIP DIS: Node Discovered during IsActiveOperationalParse
09-02 05:53:34.666  1061  3877 E CHIP DIS: OnOperationalNodeResolved - Let's ReArmTimer
09-02 05:53:34.666  1061  3877 I CHIP DIS: Discovery does not require any more timeouts
09-02 05:53:34.666  1061  3877 E CHIP DIS: Resolver inactive
09-02 05:53:34.866  1061  3877 I CHIP DIS: Checking node lookup status after 200 ms
09-02 05:53:49.668  1061  3877 I CHIP DIS: Checking node lookup status after 15003 ms
09-02 05:53:49.668  1061  3877 E CHIP DIS: Operational discovery failed for 0x024E7EEE8038E58B: 32
09-02 05:53:49.668  1061  3877 I CHIP CTL: OperationalDeviceProxy[A4575F13C9B1C48B:024E7EEE8038E58B]: State change 2 --> 1
09-02 05:53:49.668  1061  3877 I CHIP ZCL: Device connection failure
09-02 05:53:49.669  1061  3877 I CHIP ZCL: Default Failure Response: CHIP Error 0x00000032: Timeout
```
The min-mDNS is getting the records correctly, while "IsActiveOperationalParse". but when it comes to run "OnOperationalNodeResolved"  it goes directly to "ReArmTimer();" 
as if, the mActiveLookups.begin() = mActiveLookups.end()
```
auto it = mActiveLookups.begin();
    while (it != mActiveLookups.end())
```
These are the extra debug logs that I added:
```
void MinMdnsResolver::AdvancePendingResolverStates()
{
    for (IncrementalResolver * resolver = mPacketParser.ResolverBegin(); resolver != mPacketParser.ResolverEnd(); resolver++)
    {
        if (!resolver->IsActive())
        {
#if CHIP_MINMDNS_HIGH_VERBOSITY
            ChipLogError(Discovery, "Resolver inactive");
#endif
            continue;
        }

        IncrementalResolver::RequiredInformationFlags missing = resolver->GetMissingRequiredInformation();

        if (missing.Has(IncrementalResolver::RequiredInformationBitFlags::kIpAddress))
        {
#if CHIP_MINMDNS_HIGH_VERBOSITY
            ChipLogError(Discovery, "Missing information");
#endif
            ScheduleIpAddressResolve(resolver->GetTargetHostName());
            continue;
        }

        if (missing.HasAny())
        {
            // Expect either IP missing (ask for it) or done. Anything else is not handled
            ChipLogError(Discovery, "Unexpected state: cannot advance resolver with missing information");
            resolver->ResetToInactive();
            continue;
        }

        // SUCCESS. Call the delegates
        if (resolver->IsActiveCommissionParse())
        {
[... 100 lines cut ...]

This is one example of how the process works correctly:
```
09-02 05:52:37.460  1061  1608 I CHIP API: aceChip_clusterSendCommand
09-02 05:52:37.460  1061  1608 I CHIP API: aceChip_clusterSendCommand() confirmation_id=17
09-02 05:52:37.462  1061  1608 I CHIP CSM: FindOrEstablishSession: PeerId = A4575F13C9B1C48B:024E7EED29FAB08B
09-02 05:52:37.462  1061  1608 I CHIP CSM: FindOrEstablishSession: No existing OperationalDeviceProxy instance found
09-02 05:52:37.462  1061  1608 I CHIP CTL: OperationalDeviceProxy[A4575F13C9B1C48B:024E7EED29FAB08B]: State change 1 --> 2
09-02 05:52:37.462  1061  1608 I CHIP DIS: MINMDNS: Sending query ANY/UNICAST for A4575F13C9B1C48B-024E7EED29FAB08B._matter._tcp.local
09-02 05:52:37.464  1061  3877 E CHIP DIS: Truncated responses not supported for address resolution
09-02 05:52:37.464  1061  3877 I CHIP DIS: MINMDNS: received SRV record for A4575F13C9B1C48B-024E7EED29FAB08B._matter._tcp.local
09-02 05:52:37.464  1061  3877 I CHIP DIS: MINMDNS:     Operational SRV for A4575F13C9B1C48B-024E7EED29FAB08B: 5E9515632CCD0084.local
09-02 05:52:37.464  1061  3877 E CHIP DIS: Truncated responses not supported for address resolution
09-02 05:52:37.464  1061  3877 I CHIP DIS: MINMDNS: received AAAA record for 5E9515632CCD0084.local
09-02 05:52:37.464  1061  3877 I CHIP DIS: MINMDNS:     IP address fdc5:5696:c095:1:d212:36dd:becd:a156 found for 5E9515632CCD0084.local
09-02 05:52:37.464  1061  3877 I CHIP DIS: MINMDNS: received TXT record for A4575F13C9B1C48B-024E7EED29FAB08B._matter._tcp.local
09-02 05:52:37.464  1061  3877 E CHIP DIS: Node Discovered during IsActiveOperationalParse
09-02 05:52:37.465  1061  3877 I CHIP DIS: Lookup clearing interface for non LL address
09-02 05:52:37.465  1061  3877 I CHIP DIS: UDP:[fdc5:5696:c095:1:d212:36dd:becd:a156%wlan0]:5540: new best score: 6
09-02 05:52:37.465  1061  3877 E CHIP DIS: OnOperationalNodeResolved - inside the while loop
09-02 05:52:37.465  1061  3877 I CHIP DIS: Checking node lookup status after 3 ms
09-02 05:52:37.465  1061  3877 I CHIP DIS: Keeping DNSSD lookup active
09-02 05:52:37.465  1061  3877 E CHIP DIS: OnOperationalNodeResolved - Let's ReArmTimer
```
Please advice

andy31415, 2022-09-02:
@AlanLCollins - "Truncated responses not supported for address resolution" is a bit worrisome.

beyond that, do you have the ability to packet capture the mdns traffic to see what goes over the wire? is it what you expect?

andy31415, 2022-09-02:
For `as if, the mActiveLookups.begin() = mActiveLookups.end()` ... are they? This should be debuggable. What makes you say `as if`?

andy31415, 2022-09-02:
@AlanLCollins please also describe the topology and runtime a bit more:

- what SHA is this (on what branch)
- what tool is running minMDNS?
- how is this related to avahi? Is avahi running on the same machine as the minmdns (both listening on 5353 is probably not ok, there generally should be a single mdns server)?
- what topology do you have (like Device1 - thread, device2 - boerderrouter, device3 - chiptool on wifi or something like that)?

MinMDNS is allowed to receive records while mActiveLookups is empty - minMDNS is listening for all mdns traffic and packets, so it will receive packets even when no active lookups are active. This includes receiving several replies for the same query (e.g. replies on different network cards).

Having minmdns receive packets while mActiveLookups is empty is not an error. The issue title/description needs updating.

andy31415, 2022-09-02:
I updated the issue title.
Debugging this would require seeing what `OnNodeDiscovered` actually does. More logs containing the parameters for OnNodeDiscovered would help (to see if it is actually the queried node) and maybe traffic capture (to understand truncated logs).

AlanLCollins, 2022-09-07:
updates shared in the slack channel: https://csamembers.slack.com/archives/G014G30SVV0/p1662100328321889 
The min-mDNS rejects fast responses (< 1ms) from avahi  because LookupNode has not yet pushed the attempt to mActiveLookups. 
This is from a success attempt:
```
09-02 23:04:52.583  1223  1558 I CHIP API: aceChip_clusterSendCommand
09-02 23:04:52.583  1223  1558 I CHIP API: aceChip_clusterSendCommand() confirmation_id=8
09-02 23:04:52.583  1223  1558 I CHIP CSM: FindOrEstablishSession: PeerId = A4575F13C9B1C48B:024E848F7114192A
09-02 23:04:52.583  1223  1558 I CHIP CSM: FindOrEstablishSession: No existing OperationalDeviceProxy instance found
09-02 23:04:52.583  1223  1558 I CHIP CTL: OperationalDeviceProxy[A4575F13C9B1C48B:024E848F7114192A]: State change 1 --> 2
09-02 23:04:52.583  1223  1558 I CHIP DIS: MINMDNS: Sending query ANY/UNICAST for A4575F13C9B1C48B-024E848F7114192A._matter._tcp.local
09-02 23:04:52.584  1223  1558 E CHIP DIS: LookupNode - mActiveLookups.PushBack
09-02 23:04:52.584  1223  3763 E CHIP DIS: Truncated responses not supported for address resolution
09-02 23:04:52.584  1223  3763 I CHIP DIS: MINMDNS: received SRV record for A4575F13C9B1C48B-024E848F7114192A._matter._tcp.local
09-02 23:04:52.584  1223  3763 I CHIP DIS: MINMDNS:     Operational SRV for A4575F13C9B1C48B-024E848F7114192A: 5617D0C11C41DBF3.local
09-02 23:04:52.584  1223  3763 E CHIP DIS: Truncated responses not supported for address resolution
09-02 23:04:52.584  1223  3763 I CHIP DIS: MINMDNS: received AAAA record for 5617D0C11C41DBF3.local
09-02 23:04:52.584  1223  3763 I CHIP DIS: MINMDNS:     IP address fd23:d04c:fe21:1:6229:5bc7:cb3f:858f found for 5617D0C11C41DBF3.local
09-02 23:04:52.584  1223  3763 I CHIP DIS: MINMDNS: received TXT record for A4575F13C9B1C48B-024E848F7114192A._matter._tcp.local
09-02 23:04:52.585  1223  3763 E CHIP DIS: Node Discovered during IsActiveOperationalParse
09-02 23:04:52.585  1223  3763 I CHIP DIS: Lookup clearing interface for non LL address
09-02 23:04:52.585  1223  3763 I CHIP DIS: UDP:[fd23:d04c:fe21:1:6229:5bc7:cb3f:858f%wlan0]:5540: new best score: 6
09-02 23:04:52.585  1223  3763 E CHIP DIS: OnOperationalNodeResolved - inside the while loop
```
this is from a failure:
```
09-02 23:06:19.996  1223  1558 I CHIP API: aceChip_clusterSendCommand
09-02 23:06:19.996  1223  1558 I CHIP API: aceChip_clusterSendCommand() confirmation_id=23
09-02 23:06:19.996  1223  1558 I CHIP CSM: FindOrEstablishSession: PeerId = A4575F13C9B1C48B:024E8491038845E6
09-02 23:06:19.996  1223  1558 I CHIP CSM: FindOrEstablishSession: No existing OperationalDeviceProxy instance found
09-02 23:06:19.996  1223  1558 I CHIP CTL: OperationalDeviceProxy[A4575F13C9B1C48B:024E8491038845E6]: State change 1 --> 2
09-02 23:06:19.996  1223  1558 I CHIP DIS: MINMDNS: Sending query ANY/UNICAST for A4575F13C9B1C48B-024E8491038845E6._matter._tcp.local
09-02 23:06:19.997  1223  3763 E CHIP DIS: Truncated responses not supported for address resolution
09-02 23:06:19.997  1223  3763 I CHIP DIS: MINMDNS: received SRV record for A4575F13C9B1C48B-024E8491038845E6._matter._tcp.local
09-02 23:06:19.997  1223  3763 I CHIP DIS: MINMDNS:     Operational SRV for A4575F13C9B1C48B-024E8491038845E6: 56987F42D5602A8C.local
09-02 23:06:19.997  1223  3763 E CHIP DIS: Truncated responses not supported for address resolution
09-02 23:06:19.997  1223  3763 I CHIP DIS: MINMDNS: received AAAA record for 56987F42D5602A8C.local
09-02 23:06:19.997  1223  3763 I CHIP DIS: MINMDNS:     IP address fd23:d04c:fe21:1:d50d:d370:c094:ac63 found for 56987F42D5602A8C.local
09-02 23:06:19.997  1223  3763 I CHIP DIS: MINMDNS: received TXT record for A4575F13C9B1C48B-024E8491038845E6._matter._tcp.local
09-02 23:06:19.997  1223  3763 E CHIP DIS: Node Discovered during IsActiveOperationalParse
09-02 23:06:19.997  1223  3763 E CHIP DIS: OnOperationalNodeResolved - Let's ReArmTimer
09-02 23:06:19.997  1223  3763 I CHIP DIS: Discovery does not require any more timeouts
09-02 23:06:19.997  1223  3763 E CHIP DIS: Resolver inactive
09-02 23:06:19.998  1223  1558 E CHIP DIS: LookupNode - mActiveLookups.PushBack
```

AlanLCollins, 2022-09-08:
@andy31415 , Do you see an issue with registering an active lookup before calling "Dnssd::Resolver::Instance().ResolveNodeId" ? 
If that fails to a publish an mDNS query in the IP interface, the active lookup will get automatically aged and removed.

andy31415, 2022-09-08:
@AlanLCollins  Is this doing the `mActiveLookups.PushBack(&handle);` before ResolveNodeId call in AddressResolve_DefaultImpl.cpp ?

I think reordering should be fine, what is a bit unclear to me is threading - chip has a strong assumption that all processing is in a single thread, so processing of any mdns responses should happen in the chip event/thread loop as well, so I would have not expected the ResolveNodeId to manage to process a response before mActiveLookups push to happen.

Could you figure out what threads are involved? Maybe, if this is the spot of the error, we could add some `sleep()` calls before the PushBack and see if that makes it consistently reproducible.

andy31415, 2022-09-08:
CHIP strongly assumes that only one event loop is running, there is no other synchronization.

I believe JNI has `chip::DeviceLayer::StackLock` or overall code can use `PlatformMgr().LockChipStack()` and `PlatformMgr().UnockChipStack()` if you need to call chip methods outside its main thread.

andy31415, 2022-09-12:
Closing the issue: I believe this is due to threading issue on the user of the SDK, not something the SDK can fix.

closed 2022-09-12 by andy31415