project-chip/connectedhomeip #22360: Instructions how to setup RPi with eth0 only not available

HomeACcessoryKid, 2022-09-02:
#### Problem
I would like to see some instructions how the configuration of ChipTool would be done for multiple networking layouts
The one I am after in particular is using only eth0 on a Raspberry Pi.

I found some variables like INFRA_IF_NAME, BACKBONE_IFNAME and WAN_INTERFACE but the current guides contain no hints whatsoever about what to do with them to achieve a network topology.

 #### Proposed Solution
Additional section in the guides, or at least an answer to this issue.

Many thanks in advance!

tcarmelveilleux, 2022-09-02:
It's unclear what exactly you are looking for. Could you please give a bit more context about what you are expecting that is not working? Matter SDK by default operates on all network interfaces. Are you meaning that you want to limit it to only 1 interface overall?

HomeACcessoryKid, 2022-09-03:
#### Problem
 * expected behavior
After commissioning the lighting device on ESP32 using BLE which seemed successful, the controller is able to find the device via the eth0 interface where from a topology point of view the device is situated.

 * actual behavior
Not sure anymore. Initially I thought it was using wlan0 only but I repeated my experiment and did a trace on the eth0 port. Some things do happen but the end result is still that it failed.  
With the given answer that @tcarmelveilleux gave, I will continue my debugging.

 * steps to reproduce
download and install the SiLabs controller from https://github.com/SiliconLabs/matter/blob/release_0.2.0/docs/silabs/general/ARTIFACTS.md
```
connectedhomeip/out/standalone/chip-tool pairing ble-wifi 345 MyAP mypass 20202021 3840
[1661549457.796971][1220:1225] CHIP:TOO: Device commissioning completed with success

connectedhomeip/out/standalone/chip-tool onoff toggle 345 1
[1661549506.769435][1250:1250] CHIP:TOO: Run command failure: ../../examples/chip-tool/commands/common/CHIPCommand.cpp:454: CHIP Error 0x00000032: Timeout
```

 * system configuration
RPi 3B was able to work with Thread examples, but in this experiment there is the intention to ONLY use the eth0 interface. This means the wifi sits outside the RPi

 * screenshots, images, logs, etc.
[image]
 
 #### Proposed Solution
I will keep on experimenting and close this issue.

Thanks for your quick response!

closed 2022-09-03 by HomeACcessoryKid