project-chip/connectedhomeip #22361: Facing nlassert.h : no such file or directory error

suneel-eng, 2022-09-02:
I am trying to build chip client. for this I am following [https://codelabs.developers.google.com/codelabs/chip-get-started#3](https://codelabs.developers.google.com/codelabs/chip-get-started#3) documentation.

In examples/chip-tool, when I run ```shell ninja -C out/debug``` command I am getting error as below.
[image]

I am using Ubuntu 20.04 LTS in virtual box with windows 10 as Host OS.

Please let me know, possible solutions for this.

Thanks in advance.

bzbarsky-apple, 2022-09-02:
@suneel-eng Does `third_party/nlassert/repo/include/nlassert.h` exist in your tree at toplevel?

If not, did you run the `git submodule update --init` line from the instructions?

suneel-eng, 2022-09-05:
@bzbarsky-apple , Thanks for the response.

1) Does third_party/nlassert/repo/include/nlassert.h exist in your tree at toplevel?

**include** folder doesn't exist in the repo folder. repo folder is empty.

2) If not, did you run the git submodule update --init line from the instructions?

Yes, I have followed the exact steps as per the documentation.

suneel-eng, 2022-09-05:
@bzbarsky-apple , sorry. previously I did ```shell git submodule --init```. so this error was occured.
It was resolved after running ```shell git submodule update --init```.
Thanks you.

But, even after running ```shell source third_party/connectedhomeip/scripts/activate.sh ``` command I am getting error as below.

[image]

andy31415, 2022-09-06:
Do the following:

```
rm -rf ./.environment

# Start a fresh shell. Close the current one, open a new one. `which python` should point at the system python
git submodule update --init -f --recursive
source ./scripts/bootstrap.sh
```

This should make it work.

Generally activate error means the previous bootstrap had problems or was incomplete (hence me asking for you to remove .environment and starting a new shell)

closed 2023-03-18 by stale[bot]