project-chip/connectedhomeip #22384: Missing Attributes in otasoftwareupdaterequestor read impacting step 4 and 5 in TC-SU-4.2

SaileeG, 2022-09-02:
Missing Attributes in otasoftwareupdaterequestor read impacting step 4 and 5 in TC-SU-4.2 

Missing attributes:
- update-state
- update-state-progress

```
skorgaonkar-macbookpro2:connectedhomeip skorgaonkar$ ./chip-tool otasoftwareupdaterequestor read update-state 1 0
[1662158967333] [8563:106998] CHIP: [TOO] Unknown attribute: update-state
Usage:
  ./chip-tool otasoftwareupdaterequestor read attribute-name [param1 param2 ...]

  +-------------------------------------------------------------------------------------+
  | Attributes:                                                                         |
  +-------------------------------------------------------------------------------------+
  | * default-ota-provider                                                              |
  | * update-possible                                                                   |
  | * cluster-revision                                                                  |
  +-------------------------------------------------------------------------------------+
[1662158967334] [8563:106998] CHIP: [TOO] Run command failure: examples/chip-tool/commands/common/Commands.cpp:107: Error 0x0000002F
```

```
skorgaonkar-macbookpro2:connectedhomeip skorgaonkar$ ./chip-tool otasoftwareupdaterequestor read update-state-progress 1 0
[1662159124054] [8743:110079] CHIP: [TOO] Unknown attribute: update-state-progress
Usage:
  ./chip-tool otasoftwareupdaterequestor read attribute-name [param1 param2 ...]

  +-------------------------------------------------------------------------------------+
  | Attributes:                                                                         |
  +-------------------------------------------------------------------------------------+
  | * default-ota-provider                                                              |
  | * update-possible                                                                   |
  | * cluster-revision                                                                  |
  +-------------------------------------------------------------------------------------+
[1662159124055] [8743:110079] CHIP: [TOO] Run command failure: examples/chip-tool/commands/common/Commands.cpp:107: Error 0x0000002F
skorgaonkar-macbookpro2:connectedhomeip skorgaonkar$ 

```

cjandhyala, 2022-09-03:
@SaileeG What commitID are you on , I am on SVE2 branch and See the command

[image]

andy31415, 2022-09-06:
Removing tags. Used ToT today and compiled chip-tool:

```
./out/linux-x64-chip-tool/chip-tool otasoftwareupdaterequestor read
[1662476819.124583][1720689:1720689] CHIP:TOO: Missing attribute name
Usage:
  ./out/linux-x64-chip-tool/chip-tool otasoftwareupdaterequestor read attribute-name [param1 param2 ...]

  +-------------------------------------------------------------------------------------+
  | Attributes:                                                                         |
  +-------------------------------------------------------------------------------------+
  | * default-ota-providers                                                             |
  | * update-possible                                                                   |
  | * update-state                                                                      |
  | * update-state-progress                                                             |
  | * generated-command-list                                                            |
  | * accepted-command-list                                                             |
  | * attribute-list                                                                    |
  | * feature-map                                                                       |
  | * cluster-revision                                                                  |
  +-------------------------------------------------------------------------------------+
[1662476819.124606][1720689:1720689] CHIP:TOO: Run command failure: ../../examples/chip-tool/commands/common/Commands.cpp:129: Error 0x0000002F
```

closed 2022-09-06 by SaileeG