project-chip/connectedhomeip #22386: [TC-IDM-6.2] Chip-tool returns invalid argument when subscribing to start-up event

pankore, 2022-09-03:
#### Problem
- First launch interactive mode
- Step 1: basic subscribe-event  start-up  20 100 1 0 (no errors)
- Step 9: basic subscribe-event  start-up  20 100 1 0 (returns invalid argument immediately)
- DUT, did not receive anything

Chip-tool SHA: 0962e0b82254cc2ee7c86ac37e8e3d99805422f1 on sve-2 branch

bzbarsky-apple, 2022-09-04:
@pankore Please attach (not paste) logs from chip-tool and the DUT.

shuqunx, 2022-09-04:
@bzbarsky-apple Please refer to the logs attached.
[[TC-IDM-6.2]_log.txt](https://github.com/project-chip/connectedhomeip/files/9484100/TC-IDM-6.2._log.txt)

Rajashreekalmane, 2022-09-07:
@shuqunx In IDM each step tries to verify the different scenarios, Yes I agree If you continuously execute from step 1 to 9, Then 9 will get Invalid Argument due to is-urgent flag in step 8 is set to true for the same subscribe event. 
If you try this step by configuring false for is-urgent attribute for subscribe, then step 9 will work.

bzbarsky-apple, 2022-09-07:
> Then 9 will get Invalid Argument due to is-urgent flag in step 8 is set to true for the same subscribe event.

@Rajashreekalmane but it shouldn't!

@shuqunx Thank you, the specific steps and logs there are very helpful in diagnosing this.  This was due to memory corruption in chip-tool when handling the `--is-urgent` argument, which was just fixed in #22406

closed 2022-09-07 by bzbarsky-apple