project-chip/connectedhomeip #22394: [EFR32] Commissioning fails when Apple home pod mini is present on the network

rosahay-silabs, 2022-09-05:
#### Problem
* expected behavior: 
       Commissioning should work with the presence of Apple home pod mini.
* actual behavior:
       Commissioning failing when Apple home pod mini is present on the network, but after taking the home pod mini offline the commissioning works fine.
       `<error > [DIS] Failed to reply to query: 3000001`

* steps to reproduce: Connect/Disconnect the apple home pod mini and get respective failure/success.
* system configuration: Apple home pod mini (MODEL: MY5G2LL/A, VERSION: 15.6) and EFR32MG12 + RS911x, Ubuntu PC running chip-tool.
* logs: 
   [chip-tool.01.log](https://github.com/project-chip/connectedhomeip/files/9488134/chip-tool.01.log)
   [jlink-rtt-client.01.log](https://github.com/project-chip/connectedhomeip/files/9488135/jlink-rtt-client.01.log)
   [chip-tool.02.log](https://github.com/project-chip/connectedhomeip/files/9488133/chip-tool.02.log)
   [jlink-rtt-client.02.log](https://github.com/project-chip/connectedhomeip/files/9488137/jlink-rtt-client.02.log)
 #### Proposed Solution
TBD

bzbarsky-apple, 2022-09-06:
@rosahay-silabs That's very strange.  Is the homepod related to the Matter fabric involved in any way?

rosahay-silabs, 2022-09-07:
> @rosahay-silabs That's very strange. Is the homepod related to the Matter fabric involved in any way?

Honestly, that is what I am trying to figure out. The Apple homepod is running as a thread border router, with a nanoleaf light(essentials a19 bulb). I have another apple homepod mini with me (MJ2C3HN/A), which is not causing any problem. The only difference between the two is the operation as thread router.

rosahay-silabs, 2022-09-08:
After resetting the homepod mini which was running as thread border router, the commissioning started working. So, the issue is between homepod mini running as thread border router and a MATTER Wi-Fi device being on the same subnet.

rosahay-silabs, 2022-09-08:
After resetting the homepod mini which was running as thread border router, the commissioning started working. So, the issue is between homepod mini running as thread border router and a Wi-Fi device being on the same subnet.

rosahay-silabs, 2023-03-13:
Retested this seems to be working fine now.

closed 2023-03-13 by rosahay-silabs