project-chip/connectedhomeip #22408: Building and Running CHIP Linux Examples for i.MX 8M Mini EVK

MikeYJ6, 2022-09-06:
#### Problem
 I meet a problem in the subtitle "Build the example application:"of 
"Building and Running CHIP Linux Examples for i.MX 8M Mini EVK"
( connectedhomeip/docs/guides/nxp_imx8m_linux_examples.md ).

The problem is i can not find these files : scripts/activate.sh   scripts/examples/imxlinux_example.sh   scripts/build/build_examples.py.

I want to know where is the path of  these files,  thank you for replying .

bzbarsky-apple, 2022-09-06:
Those are paths relative to the root of the connectedhomeip repo: https://github.com/project-chip/connectedhomeip/blob/master/scripts/activate.sh and https://github.com/project-chip/connectedhomeip/blob/master/scripts/examples/imxlinux_example.sh

closed 2022-09-06 by bzbarsky-apple

MikeYJ6, 2022-09-07:
Thank you for  give me the path of the files .
Can I ask anthor question?
In the subtitle "Build the example application",
there is a command :
./scripts/examples/imxlinux_example.sh examples/all-clusters-app/linux examples/all-clusters-app/linux/out/aarch64.

I could not find the path "examples/all-clusters-app/linux/out/aarch64" on Github, is something wrong with that?
I want to know the path of "aarch64". If you know, please tell me the path.
thank you for replying.

bzbarsky-apple, 2022-09-07:
> I could not find the path "examples/all-clusters-app/linux/out/aarch64" on Github

The last argument to that script is the directory to place the build (objdir) in.  It's not part of the source tree, and you can choose whatever directory name you want.

MikeYJ6, 2022-09-08:
Thank you for give me the answer.
I execute the scripts/activate.sh and the result is as below.

[image]

It looks like there are some errors in the step "setting up Python environment".

What do these errors mean? [it sees like my enviroment must be advised]     How can I resolve thees errors? 
And  in what file output the log 
"Setting up Python environment.....[-]Ignoring pyobjc-core: markers 'sys_platform == "darwin"' don't match your environment"?

Thank you for replying.

bzbarsky-apple, 2022-09-08:
@MikeYJ6 I have no idea.  Maybe @mspang knows?

MikeYJ6, 2022-09-12:
Hi everyone, 
Thanks a lot for help! 
As the previous statements, I met some errors when  executing the scripts/activate.sh.
I found  the file "script/requirements.txt" which is about the error log and the content is as the picture below,
[image]

Does this help to solve this problem?
How can I resolve the error, is there any idea ?

MikeYJ6, 2022-09-12:
Hi everyone,
Thanks a lot for help!
As the previous statements, I met some errors when executing the scripts/activate.sh.
I found the file "script/requirements.txt" which is about the error log and the content is as the picture below,

[image]

Does this help to solve this problem? Is there any idea ?

Thank you for replying.