project-chip/connectedhomeip #22415: Failed to create python controller, Missing vendor ID and Failed to bringup CHIPDeviceController CLI

VivekPalle, 2022-09-06:
Hi,

We are trying to work with Python controller, but failed to create the environment. Please find the below log for error details.

server@server-Vostro-3470:~/tools/esp-matter/connectedhomeip/connectedhomeip$ sudo out/python_env/bin/chip-device-ctrl
[1662476695.323661][244446:244446] CHIP:CTL: Setting attestation nonce to random value
[1662476695.323714][244446:244446] CHIP:CTL: Setting CSR nonce to random value
[1662476695.325461][244446:244446] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-VeD5M2)
[1662476695.325560][244446:244446] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1662476695.325568][244446:244446] CHIP:DL: NVS set: chip-counters/reboot-count = 66 (0x42)
[1662476695.325697][244446:244446] CHIP:DL: Got Ethernet interface: enp1s0
[1662476695.325780][244446:244446] CHIP:DL: Found the primary Ethernet interface:enp1s0
[1662476695.325858][244446:244446] CHIP:DL: Got WiFi interface: wlp2s0
[1662476695.325948][244446:244446] CHIP:DL: Found the primary WiFi interface:wlp2s0
2022-09-06 20:34:55 server-Vostro-3470 root[244446] CRITICAL Loading configuration from /tmp/chip-device-ctrl-storage.json...
2022-09-06 20:34:55 server-Vostro-3470 chip.TS[244446] INFO Last Known Good Time: 2022-09-06T20:14:55
2022-09-06 20:34:55 server-Vostro-3470 root[244446] INFO SetSdkKey: g/gcc = b'\x88\x13\x00\x00'
2022-09-06 20:34:55 server-Vostro-3470 root[244446] INFO SetSdkKey: g/gdc = b'\x88\x13\x00\x00'
2022-09-06 20:34:55 server-Vostro-3470 chip.ZCL[244446] INFO Using ZAP configuration...
2022-09-06 20:34:55 server-Vostro-3470 chip.DL[244446] ERROR MDNS failed to join multicast group on enp1s0 for address type IPv4: ../../src/inet/UDPEndPointImplSockets.cpp:772: Inet Error 0x00000110: Address not found
2022-09-06 20:34:55 server-Vostro-3470 chip.IN[244446] INFO CASE Server enabling CASE session setups
2022-09-06 20:34:55 server-Vostro-3470 chip.DIS[244446] INFO Updating services using commissioning mode 0
2022-09-06 20:34:55 server-Vostro-3470 chip.DL[244446] ERROR MDNS failed to join multicast group on enp1s0 for address type IPv4: ../../src/inet/UDPEndPointImplSockets.cpp:772: Inet Error 0x00000110: Address not found
2022-09-06 20:34:55 server-Vostro-3470 chip.DIS[244446] INFO CHIP minimal mDNS started advertising.
2022-09-06 20:34:55 server-Vostro-3470 chip.DIS[244446] INFO Advertise commission parameter vendorID=65521 productID=32769 discriminator=3840/15
2022-09-06 20:34:55 server-Vostro-3470 chip.DIS[244446] INFO CHIP minimal mDNS configured as 'Commissioner device'.
2022-09-06 20:34:55 server-Vostro-3470 chip.DIS[244446] INFO mDNS service published: _matterd._udp
__init__() missing 1 required positional argument: 'vendorId'
Failed to bringup CHIPDeviceController CLI
Aborted
server@server-Vostro-3470:~/tools/esp-matter/connectedhomeip/connectedhomeip$ 

Thanks.

andy31415, 2022-09-06:
I do not think chip-device-ctrl is a supported interface anymore. We are looking to support chip-repl along side chip-tool.

andy31415, 2022-09-06:
Accepting patches in case of life support because this will not change core functionality, only python.

closed 2023-05-04 by stale[bot]