project-chip/connectedhomeip #22416: [esp32]:  [Automation]Failed to read serverClusters attribute with ESP32

yufengwangca, 2022-09-06:
#### Problem
Failed to conduct commissioning against M5 in our automation test

09-01 13:22:32.810  1582  1582 E SetupDeviceViewModel: Commissioning failed with state Operation failed. [CONTEXT service_id=336 ]
09-01 13:22:32.810  1582  1582 E SetupDeviceViewModel: m.dvk: Failed to read serverClusters attribute

It looks the root cause is the M5 run out of memory to allocate pkt to send BLE indication during service discovery phase.

The M5 has completed the PASE.
<2022-09-01 10:22:23.575041> NDM-0: �[0;32mI (162312) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 0 �[0m
<2022-09-01 10:22:23.576559> NDM-0: �[0;32mI (162322) chip[SC]: SecureSession[0x3ffc87c8]: Moving from state 'kEstablishing' --> 'kActive'�[0m
<2022-09-01 10:22:23.596899> NDM-0: �[0;32mI (162332) chip[SVR]: Commissioning completed session establishment step�[0m

The M5 start mDNS advertising for operational discovery
<2022-09-01 10:22:23.598199> NDM-0: �[0;32mI (162332) chip[DIS]: Updating services using commissioning mode 0�[0m
<2022-09-01 10:22:23.599207> NDM-0: �[0;32mI (162342) chip[DIS]: CHIP minimal mDNS started advertising.�[0m

The M5 receive the IM read request from commissioner
<2022-09-01 10:22:24.261880> NDM-0: �[0;32mI (163002) chip[EM]: Received message of type 0x2 with protocolId (0, 1) and MessageCounter:82885368 on exchange 1727r�[0
<2022-09-01 10:22:24.262727> NDM-0: m
<2022-09-01 10:22:24.288915> NDM-0: �[0;32mI (163022) chip[IN]: Prepared secure message 0x3ffe7474 to 0xFFFFFFFB00000000 (0)  of type 0x5 and protocolId (0, 1) on e
<2022-09-01 10:22:24.290207> NDM-0: xchange 1727r with MessageCounter:25318277.�[0m
<2022-09-01 10:22:24.309576> NDM-0: �[0;32mI (163042) chip[IN]: Sending encrypted msg 0x3ffe7474 with MessageCounter:25318277 to 0xFFFFFFFB00000000 (0) at monotonic
<2022-09-01 10:22:24.310964> NDM-0:  time: 00000000000279DE msec�[0m
<2022-09-01 10:22:24.312273> NDM-0: �[0;32mI (163052) NimBLE: GATT procedure initiated: indicate; �[0m
<2022-09-01 10:22:24.313463> NDM-0: �[0;32mI (163052) NimBLE: att_handle=14
<2022-09-01 10:22:24.314535> NDM-0: �[0m
<2022-09-01 10:22:24.332795> NDM-0: �[0;32mI (163072) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 0 �[0m
<2022-09-01 10:22:24.334056> NDM-0: �[0;32mI (163072) NimBLE: GATT procedure initiated: indicate; �[0m
<2022-09-01 10:22:24.335031> NDM-0: �[0;32mI (163082) NimBLE: att_handle=14
<2022-09-01 10:22:24.336193> NDM-0: �[0m
<2022-09-01 10:22:24.355953> NDM-0: �[0;32mI (163092) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 0 �[0m
<2022-09-01 10:22:24.357269> NDM-0: �[0;32mI (163092) NimBLE: GATT procedure initiated: indicate; �[0m
<2022-09-01 10:22:24.358337> NDM-0: �[0;32mI (163102) NimBLE: att_handle=14
<2022-09-01 10:22:24.359275> NDM-0: �[0m
<2022-09-01 10:22:24.377801> NDM-0: �[0;32mI (163112) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 0 �[0m
<2022-09-01 10:22:24.379083> NDM-0: �[0;32mI (163122) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 6 �[0m

The M5 failed to send BLE indication because it failed to allocate pkt with error BLE_HS_ENOMEM
<2022-09-01 10:22:24.400291> NDM-0: �[0;31mE (163122) chip[DL]: ble_gattc_indicate_custom() failed: Error CHIP:0x0000000B�[0m
<2022-09-01 10:22:24.401665> NDM-0: �[0;31mE (163142) chip[DL]: BLEManagerImpl::SendIndication() failed: Error CHIP:0x0000000B�[0m

 #### Proposed Solution
<suggested fix, suggested enhancement>

yufengwangca, 2022-09-06:
@dhrishi We hit commissioning error in our internal automation test against esp32, it looks we reached the limit of BLE_GATT_MAX_PROCS, could we increase this limit to increase the reliability of commissioning?  

I can provide more detailed log if needed.

dhrishi, 2022-09-07:
@yufengwangca Please check if #22446 fixes your issue

dhrishi, 2022-10-14:
@yufengwangca Can you please close this issue if you haven't encountered the erroneous scenario again with the fix mentioned above?

dhrishi, 2022-11-18:
@yufengwangca Closing this now. Feel free to reopen in case you still face the issue

closed 2022-11-18 by dhrishi