project-chip/connectedhomeip #22419: [1.0] attestation elements and signature are not available during opcert signing

chrisdecenzo, 2022-09-06:
### Reproduction steps

```bash
1.Configure Android ChipDeviceController for opcert signing callback
2.AttestationElements and Signature object are often corrupted since these are not deep copied into the CommissioningParameters object (so the memory is freed after the AttestationValidation step). 
3.The AttestationElements and Signature objects provided to the opcert signing callback are often corrupted
```

### Bug prevalence

Memory corruption if often

### GitHub hash of the SDK that was being used

master

### Platform

android

### Type

Core SDK Memory Issue

andy31415, 2022-09-06:
@chrisdecenzo - `1.Configure Android ChipDeviceController for opcert signing callback`  could you share code or commands to do this?

How does a developer reproduce the bug?

chrisdecenzo, 2022-09-12:
hi @andy31415 - in ChipDeviceController.java, assign a NOCChainIssuer (using setNOCChainIssuer).

PR #22369 (2022-09-14): Address testing feedback from opcert callbacks, and re-commissioning
#### Problem
* Bugs found during testing of new external opcert signing - https://github.com/project-chip/connectedhomeip/issues/22419
* Bug when attempting to commission an already commissioned device - https://github.com/project-chip/connectedhomeip/issues/22421

#### Change overview
* Fix typo on JNI callback lookup
* Persist attestation elements and signature so they are available during opcert signing
* Read fabric list during attribute read step of commissioning and persist the nodeId if the device is already on the com [... cut]
closed 2022-09-14 by chrisdecenzo