project-chip/connectedhomeip #22420: [1.0] Update Darwin API for 1.0 launch

woody-apple, 2022-09-06:
### Reproduction steps

```bash
No reproduction steps, just API updates.
```

### Bug prevalence

API Updates

### GitHub hash of the SDK that was being used

e535710fe92824b9ad6a902e560f763173a0bd03

### Platform

darwin

### Type

Platform Issue

andy31415, 2022-09-06:
@woody-apple could you describe what 'API Updates' actually means?
This bug seems very vague. I guess that may be by desgined as it was written and assigned exclusively to Apple engineers, however some form of description would still be nice.

bzbarsky-apple, 2022-09-06:
> could you describe what 'API Updates' actually means?

We had some internal API reviews of Matter.framework to align it with some API best practices.  This issue is about turning the results of those reviews into code changes.

woody-apple, 2022-09-15:
Closing this, as we've got this split up enough

closed 2022-09-15 by woody-apple