project-chip/connectedhomeip #22428: [1.0] Trust validation for CD signing keys missing

tcarmelveilleux, 2022-09-06:
### Reproduction steps

```bash
1. Try to add CD verifying keys with `--cd-trust-store-path PATH_TO_CERTS` on chip-tool command line
2. Verify attestation with a CD signed by a key that has its public cert in `PATH_TO_CERTS`, but does not chain to the CSA root key
3. Observe that attestation succeeds, even though the key should not have been trusted and attestation should have failed.

Only signing key certs chaining back to the approved CSA root key (see below) must be possible to add to the `CsaCdKeysTrustStore`.

-----BEGIN CERTIFICATE-----
MIICATCCAaegAwIBAgIHY3Nhcm9vdDAKBggqhkjOPQQDAjBSMQwwCgYDVQQKDAND
U0ExLDAqBgNVBAMMI01hdHRlciBDZXJ0aWZpY2F0aW9uIGFuZCBUZXN0aW5nIENB
MRQwEgYKKwYBBAGConwCAQwEQzVBMDAgFw0yMjA3MDcxOTI4MDRaGA8yMTIyMDYx
MzE5MjgwNFowUjEMMAoGA1UECgwDQ1NBMSwwKgYDVQQDDCNNYXR0ZXIgQ2VydGlm
aWNhdGlvbiBhbmQgVGVzdGluZyBDQTEUMBIGCisGAQQBgqJ8AgEMBEM1QTAwWTAT
BgcqhkjOPQIBBggqhkjOPQMBBwNCAAQ4SjrDql2+y3IP5iEdPK1IYm/3EaCkkp+t
2GD44nf/wN4fPrYzejSEe1o6BW6ocQ6Td+7t7iUXA/3ZNQEly45Io2YwZDASBgNV
HRMBAf8ECDAGAQH/AgEBMA4GA1UdDwEB/wQEAwIBBjAdBgNVHQ4EFgQUl+Rp0MUE
FMJvxwH3fpR3OQmN9qUwHwYDVR0jBBgwFoAUl+Rp0MUEFMJvxwH3fpR3OQmN9qUw
CgYIKoZIzj0EAwIDSAAwRQIgearlB0fCJ49UoJ6xwKPdlPEopCOL9jVCviODEleI
+mQCIQDvvDCKi7kvj4R4BoFS4BVZGCk4zJ84W4tfTTfu89lRbQ==
-----END CERTIFICATE-----
```
```

### Bug prevalence

100% of the time

### GitHub hash of the SDK that was being used

aa32ccd2433edccfb9ff5b0f22939c2873a30d1b

### Platform

core

### Type

Spec Compliance Issue

emargolis, 2022-09-15:
In addition to changes in #22504 need to add valid CD signed by official CSA CD Signing Certificate.

emargolis, 2022-09-23:
Was addressed with #22685

PR #22504 (2022-09-10): Update ValidateCertificateChain() To Allow Certificate Chain Validation without Intermediate Certificate
#### Issue Being Resolved
Fixes #22428 

#### Change overview
Using this function to validate CD signing certificates added to the CD Trust Store.

Added new unit-test cases.
PR #22685 (2022-09-16): [Attestation] Updated to Use CD Signed by a Valid CSA Cert
#### Issue Being Resolved
#22428

#### Change overview
Replaced test CD with valid CD signed by official CD Signing Certificate