project-chip/connectedhomeip #22441: [Android] How do I know if a device is online?

Allen-Guof, 2022-09-07:
Now I use getConnectedDevicePointer() to determine whether the device is online. When this method returns an error. I believe this device is offline. But doing so can cause some issues.
When getConnectedDevicePointer() and it returns onConnectionFailure(). It will fails when I add a device.

<details>
<summary>log</summary>
2022-09-07 11:09:16.870 6836-6836/com.yeelight.yeelight_fluid D/CSM: FindOrEstablishSession: PeerId = [1:00000000631806B6]
2022-09-07 11:09:17.789 6836-8587/com.yeelight.yeelight_fluid D/DL: Received SendWriteRequest
2022-09-07 11:09:17.910 6836-8442/com.yeelight.yeelight_fluid D/chip.BluetoothManager: null.onCharacteristicWrite: 18ee2ef5-263d-4559-959f-4f9c429f9d11 -> 0
2022-09-07 11:09:20.051 6836-8587/com.yeelight.yeelight_fluid D/DIS: Checking node lookup status after 15002 ms
2022-09-07 11:09:20.051 6836-8587/com.yeelight.yeelight_fluid E/DIS: OperationalSessionSetup[1:0000000063180B45]: operational discovery failed: 32
2022-09-07 11:09:20.051 6836-8587/com.yeelight.yeelight_fluid D/CTL: Device connection failed. Error ../../src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:174: CHIP Error 0x00000032: Timeout
2022-09-07 11:09:20.053 6836-8587/com.yeelight.yeelight_fluid D/ChipTool: onCommissioningStatusUpdate nodeId: 1662520133   stage: FindOperational  errorCode: 50
2022-09-07 11:09:20.053 6836-8587/com.yeelight.yeelight_fluid D/CTL: Error on commissioning step 'FindOperational': '../../src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:174: CHIP Error 0x00000032: Timeout'
2022-09-07 11:09:20.053 6836-8587/com.yeelight.yeelight_fluid E/CTL: Failed to perform commissioning step 18
2022-09-07 11:09:20.053 6836-8587/com.yeelight.yeelight_fluid D/CTL: Going from commissioning step 'FindOperational' with lastErr = '../../src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:174: CHIP Error 0x00000032: Timeout' -> 'Cleanup'
2022-09-07 11:09:20.053 6836-8587/com.yeelight.yeelight_fluid D/CTL: Performing next commissioning step 'Cleanup' with completion status = '../../src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:174: CHIP Error 0x00000032: Timeout'
2022-09-07 11:09:20.055 6836-8587/com.yeelight.yeelight_fluid D/ChipTool: onCommissioningStatusUpdate nodeId: 1662520133   stage: Cleanup  errorCode: 0
2022-09-07 11:09:20.055 6836-8587/com.yeelight.yeelight_fluid D/CTL: Successfully finished commissioning step 'Cleanup'
2022-09-07 11:09:20.056 6836-8587/com.yeelight.yeelight_fluid E/ChipTool: onCommissioningComplete: nodeId: 1662520133  errorCode: 50
2022-09-07 11:09:20.056 6836-8587/com.yeelight.yeelight_fluid D/ChipTool: onCommissioningComplete failed
2022-09-07 11:09:20.059 6836-8587/com.yeelight.yeelight_fluid D/DIS: Checking node lookup status after 13185 ms
2022-09-07 11:09:20.075 6836-8500/com.yeelight.yeelight_fluid I/flutter: 🔴 DeviceConnectPage - SEVERE - 2022-09-07 11:09:20.074509:
     device connect error: PlatformException(-1, device onCommissioningComplete error, null, null)
2022-09-07 11:09:20.409 6836-8442/com.yeelight.yeelight_fluid D/chip.BluetoothManager: null.onCharacteristicChanged: 18ee2ef5-263d-4559-959f-4f9c429f9d12
2022-09-07 11:09:21.876 6836-8587/com.yeelight.yeelight_fluid D/DIS: Checking node lookup status after 15002 ms
2022-09-07 11:09:21.876 6836-8587/com.yeelight.yeelight_fluid E/DIS: OperationalSessionSetup[1:00000000631806B6]: operational discovery failed: 32
2022-09-07 11:09:21.886 6836-8587/com.yeelight.yeelight_fluid E/ChipClient: Unable to get connected device with nodeId 1662518966
    chip.devicecontroller.ChipDeviceControllerException: ../../src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:174: CHIP Error 0x00000032: Timeout
2022-09-07 11:09:21.890 6836-6836/com.yeelight.yeelight_fluid W/System.err: java.lang.IllegalStateException: Unable to get connected device with nodeId 1662518966
2022-09-07 11:09:21.891 6836-6836/com.yeelight.yeelight_fluid W/System.err:     at com.yeelight.yeelight_fluid.matter.ChipClient$getConnectedDevicePointer$2$1.onConnectionFailure(ChipClient.kt:80)
2022-09-07 11:09:21.894 6836-8587/com.yeelight.yeelight_fluid E/ChipClient: Unable to get connected device with nodeId 1662518966
    chip.devicecontroller.ChipDeviceControllerException: ../../src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:174: CHIP Error 0x00000032: Timeout
2022-09-07 11:09:21.898 6836-6836/com.yeelight.yeelight_fluid W/System.err: java.lang.IllegalStateException: Unable to get connected device with nodeId 1662518966
2022-09-07 11:09:21.899 6836-6836/com.yeelight.yeelight_fluid W/System.err:     at com.yeelight.yeelight_fluid.matter.ChipClient$getConnectedDevicePointer$2$1.onConnectionFailure(ChipClient.kt:80)
</details>

Allen-Guof, 2022-10-20:
@bzbarsky-apple Hi, any update? Hope get some guidance.

bzbarsky-apple, 2022-10-20:
@Allen-Guof You want someone familiar with the Android bits. That's not me.

Allen-Guof, 2022-10-21:
@bzbarsky-apple OK, although no progress. But still thank you.

123yangu, 2022-12-06:
also ran into this problem. I saw that the API has no offline push method for related listening devices,@Allen-Guof Get getConnectedDevicePointer(). Are there any other methods?

Allen-Guof, 2022-12-07:
@123yangu I implemented mDNS discovering myself. Then compare the ${compressFabricId-deviceId} with the mDNS service name.
Discover code:
```
    nsdManager.discoverServices(
      "_matter._tcp", NsdManager.PROTOCOL_DNS_SD,
      discoveryListener
    )
```

123yangu, 2022-12-08:
> @123yangu I implemented mDNS discovering myself. Then compare the ${compressFabricId-deviceId} with the mDNS service name. Discover code:
> 
> ```
>     nsdManager.discoverServices(
>       "_matter._tcp", NsdManager.PROTOCOL_DNS_SD,
>       discoveryListener
>     )
> ```

Thanks, I'll try

123yangu, 2022-12-08:
> @123yangu I implemented mDNS discovering myself. Then compare the ${compressFabricId-deviceId} with the mDNS service name. Discover code:
> 
> ```
>     nsdManager.discoverServices(
>       "_matter._tcp", NsdManager.PROTOCOL_DNS_SD,
>       discoveryListener
>     )
> ```

@Allen-Guof   I want to know whether the material device is online or offline, but you can only know whether it is online at initialization time. If the material device status changes later, you can't know

123yangu, 2022-12-08:
I want to know the online/offline status of the device dynamically

yunhanw-google, 2023-02-08:
you may create the IM subscription against the device, if the device offline, you will get the notification.

caipiblack, 2023-02-23:
I was also asking such feature here: https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/5322

To know if a device seems offline we just have to use the subscription. In case of timeout the device is possibly not reachable (offline)

The problem is to know when a device **is online**.

The detection of online devices can be based on mDNS:

- When the device appear online, the device (or the BR) transmit a mDNS packet in multicast.
- When we receive a mDNS packet from a device, if the device was offline, we can start a subscription to know if the device is online and be notified about liveness status of the device

But, do not forget:

- mDNS is UDP based (no guarantee of delivery)
- mDNS works with caches. It is possible to receive a mDNS message from a device that is offline if someone in the network send a mDNS query.

closed 2023-07-27 by yunhanw-google