project-chip/connectedhomeip #22445: [SVE-2] Android matter pairing failed with errorcode 50

smokinghot, 2022-09-07:
### Reproduction steps

```bash
1.Compiled chip-tool on android platform[arm64-v8a]
2.use "provision chip device with thread"
3.Scan code and enter thread info to pair device
4.Paired fail all the time.

Please check the log below.

2022-09-07 11:56:30.845 11919-11932/com.google.chip.chiptool D/chip.BluetoothManager: SiLabs-Light.onMtuChanged: connecting to CHIP device
2022-09-07 11:56:30.850 11919-11919/com.google.chip.chiptool I/DeviceProvisioningFragment: showMessage:Pairing
2022-09-07 11:56:30.866 11919-11919/com.google.chip.chiptool D/ChipDeviceController: Bluetooth connection added with ID: 1
2022-09-07 11:56:30.866 11919-11919/com.google.chip.chiptool D/ChipDeviceController: Pairing device with ID: 25
2022-09-07 11:56:30.866 11919-11919/com.google.chip.chiptool D/CTL: pairDevice() called with device ID, connection object, and pincode
2022-09-07 11:56:30.867 11919-11919/com.google.chip.chiptool D/IN: BleConnectionComplete: endPoint 0x7247df7a08
2022-09-07 11:56:30.867 11919-11919/com.google.chip.chiptool D/DL: Received GetMTU
2022-09-07 11:56:30.867 11919-11919/com.google.chip.chiptool D/AndroidBleManager: Android Manufacturer: (Google)
2022-09-07 11:56:30.867 11919-11919/com.google.chip.chiptool D/AndroidBleManager: Android Model: (Pixel 3a)
2022-09-07 11:56:30.868 11919-11919/com.google.chip.chiptool D/DL: Received SendWriteRequest
2022-09-07 11:56:30.871 11919-11919/com.google.chip.chiptool D/IN: SecureSession[0x74889fc8f0]: Allocated Type:1 LSID:31976
2022-09-07 11:56:30.871 11919-11919/com.google.chip.chiptool D/SC: Assigned local session key ID 31976
2022-09-07 11:56:30.872 11919-11919/com.google.chip.chiptool D/SC: Including MRP parameters in PBKDF param request
2022-09-07 11:56:30.872 11919-11919/com.google.chip.chiptool D/IN: Prepared unauthenticated message 0x7fe0b8f238 to 0x0000000000000000 (0)  of type 0x20 and protocolId (0, 0) on exchange 28195i with MessageCounter:175716039.
2022-09-07 11:56:30.872 11919-11919/com.google.chip.chiptool D/IN: Sending unauthenticated msg 0x7fe0b8f238 with MessageCounter:175716039 to 0x0000000000000000 at monotonic time: 0000000007D49C1C msec
2022-09-07 11:56:30.872 11919-11919/com.google.chip.chiptool D/IN: Message appended to BLE send queue
2022-09-07 11:56:30.872 11919-11919/com.google.chip.chiptool D/SC: Sent PBKDF param request
2022-09-07 11:56:30.872 11919-11919/com.google.chip.chiptool D/CTL: Setting failsafe timer from parameters
2022-09-07 11:56:30.872 11919-11919/com.google.chip.chiptool D/CTL: Setting thread operational dataset from parameters
2022-09-07 11:56:30.872 11919-11919/com.google.chip.chiptool D/CTL: Setting attempt thread scan from parameters
2022-09-07 11:56:30.872 11919-11919/com.google.chip.chiptool D/CTL: Setting attempt wifi scan from parameters
2022-09-07 11:56:30.872 11919-11919/com.google.chip.chiptool D/CTL: Setting attestation nonce from parameters
2022-09-07 11:56:30.872 11919-11919/com.google.chip.chiptool D/CTL: Setting CSR nonce from parameters
2022-09-07 11:56:30.872 11919-11919/com.google.chip.chiptool D/CTL: Setting PASE-only commissioning from parameters
2022-09-07 11:56:30.872 11919-11919/com.google.chip.chiptool D/CTL: Commission called for node ID 0x0000000000000019
2022-09-07 11:56:30.917 11919-11932/com.google.chip.chiptool D/BluetoothGatt: onConnectionUpdated() - Device=00:0D:6F:20:B2:AF interval=36 latency=0 timeout=500 status=0
2022-09-07 11:56:30.965 11919-11932/com.google.chip.chiptool D/chip.BluetoothManager: SiLabs-Light.onCharacteristicWrite: 18ee2ef5-263d-4559-959f-4f9c429f9d11 -> 0
2022-09-07 11:56:30.966 11919-11932/com.google.chip.chiptool D/DL: Received SubscribeCharacteristic
2022-09-07 11:56:30.967 11919-11932/com.google.chip.chiptool D/BluetoothGatt: setCharacteristicNotification() - uuid: 18ee2ef5-263d-4559-959f-4f9c429f9d12 enable: true
2022-09-07 11:56:31.057 11919-11932/com.google.chip.chiptool D/chip.BluetoothManager: SiLabs-Light.onDescriptorWrite: 00002902-0000-1000-8000-00805f9b34fb -> 0
2022-09-07 11:56:31.057 11919-11932/com.google.chip.chiptool D/BLE: subscribe complete, ep = 0x7247df7a08
2022-09-07 11:56:31.995 1115-1217/? D/VSC: @ 501573.790: [WO] Proposed rotation: 0, flat angle threshold: 65
2022-09-07 11:56:31.996 1115-1217/? D/VSC: @ 501573.790: [WO] orientation angle 12, orientation 0
2022-09-07 11:56:31.996 1115-1217/? D/VSC: @ 501573.790: [WO] Gyro angle integrated: 25.131652
2022-09-07 11:56:31.996 1115-1217/? D/VSC: @ 501573.790: [WO] Gyro angle integrated: 25.131652
2022-09-07 11:56:32.360 1115-1217/? I/CHRE: @ 501574.155: [AR] inconsistent: 41.176471
2022-09-07 11:56:32.508 1770-3534/? W/InputManager-JNI: Input channel object '369520d Toast (client)' was disposed without first being removed with the input manager!
2022-09-07 11:56:32.517 1770-3534/? W/NotificationService: Toast already killed. pkg=com.google.chip.chiptool token=android.os.BinderProxy@210f2f7
2022-09-07 11:56:32.800 1770-2131/? D/ConnectivityService: NetReassign [no changes]
2022-09-07 11:56:33.700 2886-11876/? I/NearbyDiscovery: loadFastPairModule: checkFeaturesAreAvailable:0 [CONTEXT service_id=265 ]
2022-09-07 11:56:33.700 2886-11876/? I/NearbyDiscovery: loadFastPairModule: FastPair module is downloaded. [CONTEXT service_id=265 ]
2022-09-07 11:56:34.074 1115-1217/? D/VSC: @ 501575.870: [WO] Proposed rotation: 0, flat angle threshold: 65
2022-09-07 11:56:34.074 1115-1217/? D/VSC: @ 501575.870: [WO] orientation angle 16, orientation 0
2022-09-07 11:56:34.074 1115-1217/? D/VSC: @ 501575.870: [WO] Gyro angle integrated: 13.268837
2022-09-07 11:56:34.074 1115-1217/? D/VSC: @ 501575.870: [WO] Gyro angle integrated: 13.268837
2022-09-07 11:56:34.655 2886-3297/? I/NearbyDiscovery: FastPairCache: Not triggering, because RSSI -61 is below the threshold of -52 [CONTEXT service_id=265 ]
2022-09-07 11:56:35.097 1770-3144/? W/InputManager-JNI: Input channel object 'd4653cd Toast (client)' was disposed without first being removed with the input manager!
2022-09-07 11:56:35.107 1770-3144/? W/NotificationService: Toast already killed. pkg=com.google.chip.chiptool token=android.os.BinderProxy@1272f85
2022-09-07 11:56:35.582 1115-1217/? I/CHRE: @ 501577.378: [AR] still: 38.431373
2022-09-07 11:56:35.833 1770-2131/? D/ConnectivityService: NetReassign [no changes]
2022-09-07 11:56:36.071 1115-1217/? D/VSC: @ 501577.863: [WO] Proposed rotation: 0, flat angle threshold: 65
2022-09-07 11:56:36.072 1115-1217/? D/VSC: @ 501577.863: [WO] orientation angle 12, orientation 0
2022-09-07 11:56:36.072 1115-1217/? D/VSC: @ 501577.863: [WO] Gyro angle integrated: 9.497058
2022-09-07 11:56:36.072 1115-1217/? D/VSC: @ 501577.863: [WO] Gyro angle integrated: 9.497058
2022-09-07 11:56:36.226 2351-19361/? D/NetworkMonitor/108: isDataStall: result=2, tcp packets received=0, latest tcp fail rate=100, consecutive dns timeout count=0
2022-09-07 11:56:36.226 2351-19361/? D/NetworkMonitor/108: Suspecting data stall, reevaluate
2022-09-07 11:56:36.226 1770-3144/? D/ConnectivityService: Data stall detected with methods: 2
2022-09-07 11:56:36.233 2351-12044/? D/NetworkMonitor/108: Validation disabled.
2022-09-07 11:56:36.241 1770-2131/? D/ConnectivityService: [108 WIFI] validation passed
2022-09-07 11:56:37.702 1770-3534/? W/InputManager-JNI: Input channel object 'bae960b Toast (client)' was disposed without first being removed with the input manager!
[... 291 lines cut ...]
2022-09-07 11:57:00.161 1115-1217/? D/VSC: @ 501601.956: [WO] Gyro angle integrated: 10.896112
2022-09-07 11:57:00.162 1115-1217/? D/VSC: @ 501601.956: [WO] Gyro angle integrated: 10.896112
2022-09-07 11:57:00.893 11919-12029/com.google.chip.chiptool E/SC: PASESession timed out while waiting for a response from the peer. Expected message type was 33
2022-09-07 11:57:00.893 11919-12029/com.google.chip.chiptool D/IN: SecureSession[0x74889fc8f0]: Released - Type:1 LSID:31976
2022-09-07 11:57:00.894 11919-12029/com.google.chip.chiptool D/DeviceProvisioningFragment: Pairing status update: 1
2022-09-07 11:57:00.895 11919-12029/com.google.chip.chiptool D/DIS: Closing all BLE connections
2022-09-07 11:57:00.895 11919-12029/com.google.chip.chiptool D/DeviceProvisioningFragment: onPairingComplete: 50
2022-09-07 11:57:00.900 11919-11919/com.google.chip.chiptool I/DeviceProvisioningFragment: showMessage:Pairing failed
```

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

chiptool

### Platform

android

### Type

Platform Issue

closed 2023-03-18 by stale[bot]

edWin-m, 2023-04-04:
@smokinghot I also have a device that produces this error. Do you know the meaning of that error code?
@bzbarsky-apple  Is there any documentation that has the list of error codes?

edWin-m, 2023-04-06:
@bzbarsky-apple thanks