project-chip/connectedhomeip #22447: Multicast binding not working.

jadhavrohit924, 2022-09-07:
###Problem
* Group multicast fails when we bind group on light app to the switch app
* Steps followed:

- Binded light and light-switch app to control light using swich.
- Created group on light app and binded to light-switch using groupId.
- Entering on/off commands don’t send any multicast to light.

closed 2022-09-07 by jadhavrohit924

jadhavrohit924, 2022-09-07:
@rosahay-silabs we need to send `chip-tool tests TestGroupDemoConfig` command to switch also. Then it will work.

somu1710, 2022-09-08:
Hi @jadhavrohit924 could u please attach the logs here it will helpful for us.The command you provided `chip-tool tests TestGroupDemoConfig` is not getting success response .It  shows run command failure in my set up.Actually for my set up if I press the button on nordic board ,signal sent to the switch app and not to the light-app  .while read onoff it shows false response.could you eloborate the steps to get success.

jadhavrohit924, 2022-09-09:
Steps for multicase binding:

* Node Id of switch used during commissioning is 0x7283 (29315 in decimal)
* Node Id of light1 used during commissioning is 0x5164 (20836 in decimal)

**Binding switch and light app**

* ``chip-tool accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [ 112233, 29315 ], "targets": null}]' 0x5164 0x0``
* ``chip-tool     binding write binding '[{"fabricIndex": 1, "node":20836, "endpoint":1, "cluster":6}]' 0x7283 0x1``

**Add groups to switch and light**

Below command (similar to [this](https://github.com/project-chip/connectedhomeip/issues/20979#issuecomment-1190351414) will add group to switch and light with group id 257

* ``chip-tool tests TestGroupDemoConfig --nodeId 29315``
* ``chip-tool tests TestGroupDemoConfig --nodeId 20836``
* ``chip-tool binding write binding '[{"fabricIndex": 1, "group": 257}]' 0x7283 0x1``

Logs

[chiptool-logs.txt](https://github.com/project-chip/connectedhomeip/files/9534532/chiptool-logs.txt)
[light-logs.txt](https://github.com/project-chip/connectedhomeip/files/9534533/light-logs.txt)
[switch-logs.txt](https://github.com/project-chip/connectedhomeip/files/9534534/switch-logs.txt)