project-chip/connectedhomeip #22454: [BUG] chip-tool interactive mode silently ignores --commissioner-nodeid argument

bluebin14, 2022-09-07:
### Reproduction steps

```bash
./chip-tool interactive start

>>> accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [1234], "targets": null}]' 1 0

>>> accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": null}]' 1 0 --commissioner-nodeid 1234
```

### Bug prevalence

always

### GitHub hash of the SDK that was being used

latest

### Platform

darwin, raspi

### Anything else?

The write command fails in interactive mode with unsupported access while in non interactive mode it works.

bzbarsky-apple, 2022-09-07:
Per slack discussion, it seems like the only way to make this work is to have commissioners addressed by the pair (fabric identity string, local node id), not just the fabric identity, because we need to have multiple different NOCs around for this to work.  @vivien-apple do you see any obvious issues with this approach?

And then using a non-ephemeral listen port will not work if multiple node ids are used with the same identity, but we really should not be using the non-ephemeral listen port case anyway.....

vivien-apple, 2022-09-08:
So per my understanding (and ignoring the non-ephemeral listen port), there are 2 issues:

* We are initialising alpha, beta, gamma, commissioners at startup **only** and so https://github.com/project-chip/connectedhomeip/blob/master/examples/chip-tool/commands/common/CHIPCommand.cpp#L398 does not give take into account the provided `--commissioner-nodeid 1234`

* As you said, in this case we need to address commissioners by the pair (fabric identity string, local node id).

**Solution 1:**
A simple and lazy way to fix that would be to keep `alpha` to what pre-exists today and add a few additional identities such as `alpha1`, `alpha2`, `alpha3` mapping to known node id (such as `0xa1111111` for alpha1, `0xa2222222` for alpha2, etc..).
The same logic beeing applied for `beta`, `gamma` (with `0xb11111111`, `0xb2222222`, ...)

It carries the advantages that debug log from testers would tell us which identity is used when message are sent to a given commissioner as it will be easy (at least for folks reading the logs) to map between the commissioner node id and what `chip-tool` uses.
It is also quick and simple to implement.

But it has the disadvantage of not providing a full support for `commissioner-node-id` obviously - even if I suspect that it may covers a very large amount of needs...

**Solution 2:**
The other solution, is obviously to address commissioners by a pair.
Ideally we would generate NOCs dynamically to not pollute the log too much when `chip-tool` starts (it is already annoying with the default identities fwiw).

The only downside of this one is just the fact that it is more work but I don't think that it too much work too.

Solution 1 && Solution 2 can be merged together to make it both flexible and convenient.

bluebin14, 2022-09-08:
For now could we use minimal invasive surgery by not creating a separate controller instance when commissioner-node-id is specified but instead replace the existing one exactly like the non-interactive mode does. There is no use case for simultaneously active controllers on same fabric yet.

bzbarsky-apple, 2022-09-08:
> We are initialising alpha, beta, gamma, commissioners at startup only

That's not actually true anymore.  The only thing we initialize at startup is kIdentityNull.  All other commissioners are initialized on an as-needed basis; see `CHIPCommand::EnsureCommissionerForIdentity`.  That was changed to support `--commissioner-name` in interactive mode, where a command might suddenly use a commissioner we did not know we might need at startup.

> but instead replace the existing one

So shut down the existing one and start up a new one?  It's not clear to me this is actually less invasive, fwiw...

bluebin14, 2022-09-09:
It would mimic the existing behavior in non interactive mode, would not overcomplicate storage and would allow stress test that does the nodeid change many times.

PR #22496 (2022-09-09): Fix --commissioner-nodeid handling in interactive mode.
1. Instead of keying commissioners by just name in chip-tool, key them by the (name, commissioner-node-id) pair.

2. Allow multiple commissioners for the same fabric (as long as they have different node ids).

#### Issue Being Resolved
* Fixes #22454

#### Change overview
See above.

Tested both the steps from #22454 and `onoff toggle 1 1` followed by `onoff toggle 1 1 --commissioner-nodeid 25`: before this PR that last command succeeds, whereas after it correctly fails with an `UNSUPPORTED_ACCCESS`.
closed 2022-09-09 by andy31415