project-chip/connectedhomeip #22466: [BUG] : otasoftwareupdaterequestor read for different attribute names with end point 0/1 seeing chip error in TH logs TC-SU-4.2

revanthb28, 2022-09-07:
### Reproduction steps

```bash
Hi Team,

We tried running test steps for TC-SU-4.2

We are experiencing chip error issue after several minutes right after receiving Read Request message.

This was experienced with attribute-names DefaultOTAProviders, UpdatePossible and update-state
```

### Bug prevalence

Everytime we run

### GitHub hash of the SDK that was being used

TH2.4

### Platform

raspi

### Anything else?

[raspi-SU.log](https://github.com/project-chip/connectedhomeip/files/9521887/raspi-SU.log)

bzbarsky-apple, 2022-09-08:
> TH2.4

That's not a hash.  What's the actual hash?

> We are experiencing chip error issue

Apart from these log messages:
```
[1662579799.373991][2060:2060] CHIP:DIS: DNSSD parse error: ../../examples/ota-requestor-app/linux/third_party/connectedhomeip/src/lib/dnssd/IncrementalResolve.cpp:267: CHIP Error 0x0000002F: Invalid argument
```
what is the actual erroneous behavior you are seeing @revanthb28 ?

bzbarsky-apple, 2022-09-13:
Thank you @SaileeG @revanthb28 .  Could you also answer my question above about what the actual erroneous behavior you're observing is?

closed 2023-03-23 by stale[bot]