project-chip/connectedhomeip #22468: [Platform] TH selects unrelated test while loading PICs

satish-hsk, 2022-09-08:
### Reproduction steps / Feature

```bash
1. Generate a pics xml by selecting a Support e.g TSTAT.C in e.g Thermostat Cluster Test Plan.xml
2. Load the generated xml in TH Project
3. Create a Test Run and observe the manual tests listed
3. Expected: Thermostat related tests e.g TC-TSTAT-3.1 & TC-TSTAT-3.2
4. Actual: The following unrelated tests are also getting picked with uploading one or more pics 
TC-ACE-1.1
TC-ACE-1.2
TC-BRBINFO-1.1
TC-BRBINFO-2.1
TC-BRBINFO-2.2
TC-BRBINFO-2.3
TC-SC-5.1
TC-SC-5.2
TC-SC-5.3
TC-SC-6.1
```

### Platform

raspi

### Platform Version(s)

2.4 (updated via scripts)

### Type

Unit tested

bzbarsky-apple, 2022-09-08:
@cjandhyala where should this actually be filed?

chrisbeach59, 2022-09-27:
@bzbarsky-apple @cjandhyala I think this should be in https://github.com/CHIP-Specifications/chip-certification-tool

bzbarsky-apple, 2022-10-06:
I filed https://github.com/CHIP-Specifications/chip-certification-tool/issues/580

closed 2022-10-06 by bzbarsky-apple