project-chip/connectedhomeip #22491: [Darwin] subscription request for multiple attributes support needed

Garik098, 2022-09-08:
### Reproduction steps / Feature

```bash
As per IDM 4.1 testcase step 10, we have to send subscription request to multiple attributes. Tried using this command - any subscribe-by-id '6,8,3' '0,1,0' 10 100 1 '1,1,1'

Getting the below error in Darwin tool,

any subscribe-by-id '6,8,3' '0,1,0' 10 100 1 '1,1,1'
[1662655642159] [1735:30756] CHIP: [TOO] InitArgs: Invalid argument cluster-id: 6,8,3
Usage:
   any subscribe-by-id cluster-id attribute-id min-interval max-interval node-id endpoint-id [--commissioner-name] [--fabric-filtered] [--keepSubscriptions]

Checked on chip tool and found it to be working fine.
```

### Platform

darwin

### Type

Manually tested with SDK

bzbarsky-apple, 2022-09-08:
> we have to send subscription request to multiple attributes.

Only if `MCORE.IDM.C.SubscribeRequest.MultipleAttributes` is true.  Which it's not for the Darwin framework.

bzbarsky-apple, 2022-09-08:
That is, we don't have framework support for this yet.  As a feature request for that, this is totally reasonable, but post-1.0.