project-chip/connectedhomeip #22512: [TC-IDM-6.2] Step 18 fails

adamb-q, 2022-09-09:
### Reproduction steps

```bash
Run TC-IDM-6.2 - on step 18 there is:

CHIP:TOO: Error: ../../examples/chip-tool/third_party/connectedhomeip/src/app/ReadClient.cpp:867: CHIP Error 0x00000032: Timeout

instead of event notification. Logs attached.
Ran on chip-tool x64 and qpg platform.
```

### Bug prevalence

Every time

### GitHub hash of the SDK that was being used

27f9fb49ac45cb3933d96ddf7b25aed6ec9b665d

### Platform

other

### Platform Version(s)

qpg

### Anything else?

[TC_IDM_6_2.1662716902_DUT.log](https://github.com/project-chip/connectedhomeip/files/9534861/TC_IDM_6_2.1662716902_DUT.log)
[TC_IDM_6_2.1662716902.log](https://github.com/project-chip/connectedhomeip/files/9534863/TC_IDM_6_2.1662716902.log)

bzbarsky-apple, 2022-09-09:
This is expected, no?  Step 16 (from the log; the step numbers in this log don't match the test plan step numbers at first glance) sends a Subscribe Request:
```
accesscontrol subscribe-event access-control-entry-changed 5 10 3 0
```
Step 17 (again from the log) then sends another subscribe request:
```
basic subscribe-event-by-id 0x000 100 1000 3 0 --event-min 1
```
which cancels the step 16 subscription on the server.  Eventually the client detects that the subscription is gone when it does not receive the expected report message.

@cjandhyala @adamb-q

bzbarsky-apple, 2022-09-09:
And note that step 18 is a bit of a red herring here.  The real key is that enough time passed since step 17 that we detected the subscription loss that triggered.

adamb-q, 2022-09-12:
@bzbarsky-apple looking at test instructions it's not expected I think? From document for step 18:

`Verify that each event record is assigned a number that is exactly 1 greater than the last created event record on that Node.
 Verify that for all the events that it has either System/Epoch timestamp or Delta System/Epoch Timestamp if this is not first event in current report`

I don't see a mismatch with test plan step numbers also?
So are the test steps wrong or am I missing something?

bzbarsky-apple, 2022-09-13:
The issue is not step 18.  The issue is step 17, which cancels the subscription from step 16, which then reports an error.  That error would be reported even if step 18 were never carried out.

> I don't see a mismatch with test plan step numbers also?

https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/interactiondatamodel.adoc#tc-idm-6-2-events-subscribe-interaction-from-th-to-dut-dut_server steps 15, 16, 17, 18 are:

15. TH sends Subscribe Request Message to DUT with EventFilterIB Node information field matching the node indicated in the path and EventMin field.
16. TH sends Subscribe Request Message to DUT with EventFilterIB Node information field matching the node indicated in the path and the event number is less than the EventMin field.
17. With an active Event subscription from TH to DUT, trigger events on DUT multiple times.
18. With an active Event subscription from TH to DUT, change attribute values on DUT to create events multiple times.
Perform a factory data reset on DUT, re-subscribe to events and continue to change attribute values on DUT to create events multiple times.

In the atttached log, I see:

```
Test Step: 15 -> TH_CR1
sent accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [1234], "targets": null}]' 2 0
```
which is not a Subscribe Request as per the test plan.  Similarly, in the log:
```
Test Step: 17 -> TH_CR1
sent basic subscribe-event-by-id 0x000 100 1000 3 0 --event-min 1
```
which sure looks like step 16 from the test plan...

@adamb-q Just to make sure, we _are_ looking at TC-IDM-6.2?

Note that it's possible that the verification step numbering does not match the test plan step numbering, if you are getting your step numbers from the verification steps spreadsheet....

adamb-q, 2022-09-13:
@bzbarsky-apple Yes, we are talking about IDM 6.2 for sure :) 
in the link you attached step numbers are confusing: 14 then 16,15, again 16, 17 and 18, in the spreadsheet they are incremental so I assume spreadsheet is better reference - so your step 17 for me is step 18 - that's right. Nevertheless - I am taking all test instructions from spreadsheet always (all the rest IDMs are passing for me) so what is the problem here? Am I not expected to get test instructions spreadsheet exclusively for IDM 6.2..? Or there is instruction error?

bzbarsky-apple, 2022-09-14:
The instructions in the spreadsheet lead to logging that will report an error for the subscription from step _16_ (in the spreadsheet's numbering), sometime after step _17_ runs, because step 17 cancels the subscription from step 16 on the server.

That error has nothing to do with step _18_.

The instructions _could_ probably be fixed to not trigger the error, but passing `--keepSubscriptions true` in step 17.  But the important part is that this error is not an actual test failure, as far as I can tell.

@cjandhyala

Rajashreekalmane, 2022-09-14:
As Boris mentioned, Seems the step 17 is cancelling the Step 16. 
If send the Step 16 with --keepsubscription true, then step 17 will not cancel the subscription, so that step 18 will pass. 

I'll retry in my setup without the --keepsubscription field and let you know the result. Because, I have tested earlier this and updated the VS command.

Rajashreekalmane, 2022-09-15:
@adamb-q 
I hope the issue you have mentioned here is, when you send write request command after activating the subscription you are facing the issue as Timeout error "CHIP:TOO: Error: ../../examples/chip-tool/third_party/connectedhomeip/src/app/ReadClient.cpp:867: CHIP Error 0x00000032: Timeout" 
For this issue you can get expected outcome by using --keepsubscription true while activating subscription
**PFA Log for the same:**
[TC-IDM-6.2 DUT log.txt](https://github.com/project-chip/connectedhomeip/files/9575374/TC-IDM-6.2.DUT.log.txt)
[TC-IDM-6.2 CNTR log.txt](https://github.com/project-chip/connectedhomeip/files/9575375/TC-IDM-6.2.CNTR.log.txt)

@bzbarsky-apple 
As you mentioned If I send the Step 16 with --keepsubscription true, then step 17 will not cancel the subscription, yes I agree but without the --keepsubscription true also we are getting the expected outcome for the test step 17, I hope the test step 17 of  test plan also expecting the same.
Here is the Log for same:
[TC-IDM-6.2 step 17 DUT.txt](https://github.com/project-chip/connectedhomeip/files/9575457/TC-IDM-6.2.step.17.DUT.txt)
[TC-IDM-6.2 step 17 CNTR.txt](https://github.com/project-chip/connectedhomeip/files/9575458/TC-IDM-6.2.step.17.CNTR.txt)

cjandhyala, 2022-09-15:
@adamb-q Per @Rajashreekalmane's comment , this issue can be closed. Pls let us know if we need to keep this issue open.

adamb-q, 2022-09-16:
yes, it can be closed, thanks

closed 2022-09-16 by adamb-q