project-chip/connectedhomeip #22530: [Platform] lighting-app example -> ModuleNotFoundError: No module named 'chip'

lboue, 2022-09-11:
### Reproduction steps / Feature

```bash
When I try to run  lighting-app example, I am getting a ModuleNotFoundError: No module named 'chip' error:

git clone https://github.com/project-chip/connectedhomeip.git

cd connectedhomeip
cd examples/lighting-app/python/lighting.py
pip install -r requirements.txt
python lighting.py

Traceback (most recent call last):
  File "/home/pi/connectedhomeip/examples/lighting-app/python/lighting.py", line 17, in <module>
    from chip.server import (
ModuleNotFoundError: No module named 'chip'
```
```

### Platform

python

### Platform Version(s)

Python 3.9.2

### Type

Manually tested with SDK

THEGOLDENPRO, 2022-10-06:
Yep, I'm getting the same exact error and I don't know where to find this module.

lboue, 2022-11-29:
Can anyone fix this error?

closed 2023-10-15 by stale[bot]