project-chip/connectedhomeip #22573: [Platform] MediaTek Genio support

pakls, 2022-09-13:
### Reproduction steps

```bash
New Platform: MediaTek Genio
```

### Platform

other

### Type

Manually tested with SDK, Platform validated

### (Optional) If manually tested please explain why this is only manually tested

Still at early stage. Tested in-house.

andy31415, 2022-09-13:
@pakls regarding `Still at early stage. Tested in-house.` - this platform was added in effectively the last week before 1.0 branch time. Hoping this will be stable enough. The late merge is a bit concerning.

closed 2022-09-13 by andy31415

pakls, 2022-09-13:
> @pakls regarding `Still at early stage. Tested in-house.` - this platform was added in effectively the last week before 1.0 branch time. Hoping this will be stable enough. The late merge is a bit concerning.

Thank you, Andy.

PR #22404 (2022-09-13): [genio] mt793x support
#### Problem
Added support of MediaTek Genio platform (mt793x)

#### Change overview
Added support of shell and lighting-app examples.

Closes #22573

#### Testing
Early support of Genio platform, shouldn't affect common code.