project-chip/connectedhomeip #22578: [BUG] [TC-I-2.3] TriggerEffect showing wrong effect variant

pankore, 2022-09-13:
### Reproduction steps

```bash
./chip-tool identify trigger-effect 0 66 1 1

Expected result:
CHIP:ZCL: RX identify:trigger effect 0x0 variant 0x42

Actual result:
CHIP:ZCL: RX identify:trigger effect 0x0 variant 0x1

Tested on linux all-clusters-app and ameba all-clusters-app
```

### Bug prevalence

Everytime i run the command

### GitHub hash of the SDK that was being used

0962e0b82254cc2ee7c86ac37e8e3d99805422f1 [sve-2 branch]

### Platform

ameba, raspi

cjandhyala, 2022-09-16:
This is an optional command and is not required for V1.0

cjandhyala, 2022-10-24:
@pankore is this issue specific to Ameba platform. this is not reproduced on raspi. Pls let us know if this is still an issue or can we close the issue ?

taoyinzhou, 2022-10-25:
@cjandhyala I also have problems running this use case. I use matter v1.0,example uses lock-app,DUT is nrf52840dk. 
The output is：
[1666684321.027357][1561:1566] CHIP:DMG: InvokeResponseMessage =
[1666684321.027397][1561:1566] CHIP:DMG: {
[1666684321.027436][1561:1566] CHIP:DMG: 	suppressResponse = false, 
[1666684321.027477][1561:1566] CHIP:DMG: 	InvokeResponseIBs =
[1666684321.027528][1561:1566] CHIP:DMG: 	[
[1666684321.027569][1561:1566] CHIP:DMG: 		InvokeResponseIB =
[1666684321.027629][1561:1566] CHIP:DMG: 		{
[1666684321.027673][1561:1566] CHIP:DMG: 			CommandStatusIB =
[1666684321.027732][1561:1566] CHIP:DMG: 			{
[1666684321.027784][1561:1566] CHIP:DMG: 				CommandPathIB =
[1666684321.027845][1561:1566] CHIP:DMG: 				{
[1666684321.027904][1561:1566] CHIP:DMG: 					EndpointId = 0x1,
[1666684321.027968][1561:1566] CHIP:DMG: 					ClusterId = 0x3,
[1666684321.028028][1561:1566] CHIP:DMG: 					CommandId = 0x40,
[1666684321.028084][1561:1566] CHIP:DMG: 				},
[1666684321.028146][1561:1566] CHIP:DMG: 				
[1666684321.028198][1561:1566] CHIP:DMG: 				StatusIB =
[1666684321.028255][1561:1566] CHIP:DMG: 				{
[1666684321.028316][1561:1566] CHIP:DMG: 					status = 0x85 (INVALID_COMMAND),
[1666684321.028375][1561:1566] CHIP:DMG: 				},
[1666684321.028437][1561:1566] CHIP:DMG: 				
[1666684321.028488][1561:1566] CHIP:DMG: 			},
[1666684321.028544][1561:1566] CHIP:DMG: 			
[1666684321.028590][1561:1566] CHIP:DMG: 		},
[1666684321.028644][1561:1566] CHIP:DMG: 		
[1666684321.028684][1561:1566] CHIP:DMG: 	],
[1666684321.028734][1561:1566] CHIP:DMG: 	
[1666684321.028774][1561:1566] CHIP:DMG: 	InteractionModelRevision = 1
[1666684321.028814][1561:1566] CHIP:DMG: },
[1666684321.028907][1561:1566] CHIP:DMG: Received Command Response Status for Endpoint=1 Cluster=0x0000_0003 Command=0x0000_0040 Status=0x85
[1666684321.028958][1561:1566] CHIP:TOO: Error: IM Error 0x00000585: General error: 0x85 (INVALID_COMMAND)

pankore, 2022-10-27:
@cjandhyala, I reproduced it on raspi platform back when I encountered this issue. Have not tested it since.

franck-apple, 2022-11-04:
@pankore please let us know if this issue is still current. If not, let's close the issue. 
In the meantime, it's obviously not a cert blocker, so I'm removing the label.

pankore, 2022-11-11:
@franck-apple, this issue is still happening on v1.0-branch. I am having trouble commissioning with chiptool on master ToT, will try again once chiptool is working.

closed 2023-05-20 by stale[bot]