project-chip/connectedhomeip #22581: [BUG] OTA not possible with nrfConnect light-switch-example and OTA-provider

benjaminnestler, 2022-09-13:
### Reproduction steps

```bash
1. Build ota-provider 

./scripts/examples/gn_build_example.sh examples/ota-provider-app/linux out/provider chip_config_network_layer_ble=false
```
2. Build chip-tool
```
./scripts/examples/gn_build_example.sh examples/chip-tool out/chip-tool
```
3. Build light-switch-app (nRFConnect @nRF5340DK) and flash
```
west build -b nrf5340dk_nrf5340_cpuapp
```
4. Form Thread-Network with OTBR
5. Commissioning the _light-switch-app_ --> nodeId = 0x100
```
sudo ./chip-tool pairing ble-thread 0x100 hex:<ot-netdata> 20202021 3840
```
6. Start the ota-provider-app with the firmware file to update
```
sudo ./chip-ota-provider-app -f /<path-to-your-matter-repo>/connectedhomeip/examples/light-switch-app/nrfconnect/build/zephyr/matter.ota
```
7. Commissioning the ota-provider --> nodeId = 1 (other nodeIds are more complicated / doesn't lead to a communication from the device to the ota-provider)
```
sudo ./chip-tool pairing onnetwork 1 20202021
```
8. Write default-ota-provider to light-switch-device
```
sudo ./chip-tool otasoftwareupdaterequestor write default-ota-providers '[{"fabricIndex": 1, "providerNodeID": 1, "endpoint": 0}]' 0x100 0 
```
9. Set ACL for Ota-provider operate-access to all clusters
```
sudo ./chip-tool accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": null}, {"fabricIndex": 1, "privilege": 3, "authMode": 2, "subjects": null, "targets": null}]' 1 0
```
10. Trigger the ota-firmware update via matter cli on physical light-switch-device
```
matter ota query
```

Output from device:
```
matter ota query
I: 1719783 [SWU]Triggered immediate OTA query for fabric: 0x1
D: 1719788 [DMG]Endpoint 0, Cluster 0x0000_002A update version to bcdfe56a
D: 1719795 [EVL]LogEvent event number: 0x0000000000000006 priority: 1, endpoint id:  0x0 cluster id: 0x0000_002A event id: 3
I: 1719809 [SWU]Stopping the Periodic Query timer
I: 1719814 [SWU]Starting the watchdog timer, timeout: 21600 seconds
D: 1719820 [SWU]Establishing session to provider node ID 0x0000000000000001 on fabric index 1
D: 1719828 [CSM]FindOrEstablishSession: PeerId = [1:0000000000000001]
D: 1719834 [CSM]FindOrEstablishSession: No existing OperationalSessionSetup instance found
D: 1719843 [CTL]OperationalSessionSetup[1:0000000000000001]: State change 1 --> 2
I: 1719850 [DIS]Resolving 0DC0FE2595D6A4FE:0000000000000001 ...
uart:~$ I: 1720051 [DIS]Checking node lookup status after 201 ms
I: 1720562 [DIS]Node ID resolved for 0DC0FE2595D6A4FE:0000000000000001
D: 1720568 [DIS]        Hostname: F4D488602D4F
D: 1720572 [DIS]        IP Address #1: 2a00:6020:4398:7d00:c14:bd5:e797:34b6
D: 1720578 [DIS]        Port: 5540
D: 1720581 [DIS]        Mrp Interval idle: 5000 ms
D: 1720584 [DIS]        Mrp Interval active: 2000 ms
D: 1720589 [DIS]        TCP Supported: 1
D: 1720592 [DIS]Lookup clearing interface for non LL address
I: 1720597 [DIS]UDP:[2a00:6020:4398:7d00:c14:bd5:e797:34b6]:5540: new best score: 5
I: 1720605 [DIS]Checking node lookup status after 754 ms
D: 1720610 [DIS]OperationalSessionSetup[1:0000000000000001]: Updating device address to UDP:[2a00:6020:4398:7d00:c14:bd5:e72
D: 1720623 [CTL]OperationalSessionSetup[1:0000000000000001]: State change 2 --> 3
D: 1720631 [IN]SecureSession[0x200047e8]: Allocated Type:2 LSID:27118
I: 1720637 [SC]Initiating session on local FabricIndex 1 from 0x0000000000000100 -> 0x0000000000000001
D: 1720780 [SC]Including MRP parameters
[... 72 lines cut ...]
[1663060399047] [8042:74588] CHIP: [EM] Found matching exchange: 11032r, Delegate: 0x104bce660
[1663060399047] [8042:74588] CHIP: [EM] Forcing tx of solitary ack for duplicate MessageCounter:247485674 on exchange 11032r
[1663060399047] [8042:74588] CHIP: [EM] Sending Standalone Ack for MessageCounter:247485674 on exchange 11032r
[1663060399047] [8042:74588] CHIP: [IN] Prepared unauthenticated message 0x30df36508 to 0x0000000000000000 (0)  of type 0x10 and protocolId (0, 0) on exchange 11032r with MessageCounter:15789319.
[1663060399047] [8042:74588] CHIP: [IN] Sending unauthenticated msg 0x30df36508 with MessageCounter:15789319 to 0x0000000000000000 at monotonic time: 00000000004ABD8F msec
[1663060409308] [8042:74588] CHIP: [SC] CASESession timed out while waiting for a response from the peer. Current state was 2
[1663060409309] [8042:74588] CHIP: [IN] CASE Session establishment failed: ../connectedhomeip/examples/ota-provider-app/linux/third_party/connectedhomeip/src/protocols/secure_channel/CASESession.cpp:290: CHIP Error 0x00000032: Timeout
[1663060409309] [8042:74588] CHIP: [IN] CASE Server enabling CASE session setups
[1663060409309] [8042:74588] CHIP: [IN] SecureSession[0x600003808000]: Released - Type:2 LSID:22827
[1663060409309] [8042:74588] CHIP: [IN] SecureSession[0x60000380c000]: Allocated Type:2 LSID:22828
[1663060409309] [8042:74588] CHIP: [SC] Allocated SecureSession (0x60000380c000) - waiting for Sigma1 msg
```
```

### Bug prevalence

Whenever I do the ota query

### GitHub hash of the SDK that was being used

5caec5a5e6f40f727d1f9d3170ecb878e664fb42

### Platform

nrf

Damian-Nordic, 2022-09-13:
What does "other nodeIds are more complicated / doesn't lead to a communication from the device to the ota-provider" exactly mean?

From the logs it seems the OTA provider is not able to deliver messages back to the OTA requestor (light switch):
```
[1663060359306] [8042:67029] CHIP: [SC] Sent Sigma2 msg
[1663060359929] [8042:74588] CHIP: [EM] Retransmitting MessageCounter:15789315 on exchange 11032r Send Cnt 1
[1663060359929] [8042:74588] CHIP: [IN] Sending unauthenticated msg 0x600000728728 with MessageCounter:15789315 to 0x0000000000000000 at monotonic time: 00000000004A24C1 msec
[1663060360486] [8042:74588] CHIP: [EM] Retransmitting MessageCounter:15789315 on exchange 11032r Send Cnt 2
[1663060360486] [8042:74588] CHIP: [IN] Sending unauthenticated msg 0x600000728728 with MessageCounter:15789315 to 0x0000000000000000 at monotonic time: 00000000004A26EE msec
[1663060361390] [8042:74588] CHIP: [EM] Retransmitting MessageCounter:15789315 on exchange 11032r Send Cnt 3
[1663060361391] [8042:74588] CHIP: [IN] Sending unauthenticated msg 0x600000728728 with MessageCounter:15789315 to 0x0000000000000000 at monotonic time: 00000000004A2A77 msec
[1663060362923] [8042:67029] CHIP: [EM] Retransmitting MessageCounter:15789315 on exchange 11032r Send Cnt 4
[1663060362923] [8042:67029] CHIP: [IN] Sending unauthenticated msg 0x600000728728 with MessageCounter:15789315 to 0x0000000000000000 at monotonic time: 00000000004A3073 msec
[1663060365315] [8042:74588] CHIP: [IN] Received a duplicate message with MessageCounter:247485674 on exchange 11032r
[1663060365315] [8042:74588] CHIP: [EM] Received message of type 0x30 with protocolId (0, 0) and MessageCounter:247485674 on exchange 11032r
```
Are you able to investigate why the device running the OTA provider application cannot talk to the light switch? You can try to manually ping the device or check the routing table if everything is configured properly.

benjaminnestler, 2022-09-14:
The device running the OTA-Provider is my Mac M1 Max.

I can discover the OTA-Provider by mDNS with discovery-Tool.
[image]
I'm able to ping the OTA-Provider from my local machine with 
```
~/workspaces/matter/chip-tool-091322 % ping6 F4D488602D4F.local           
PING6(56=40+8+8 bytes) ::1 --> ::1
16 bytes from ::1, icmp_seq=0 hlim=64 time=0.226 ms
16 bytes from ::1, icmp_seq=1 hlim=64 time=0.322 ms
16 bytes from ::1, icmp_seq=2 hlim=64 time=0.298 ms
16 bytes from ::1, icmp_seq=3 hlim=64 time=0.332 ms
16 bytes from ::1, icmp_seq=4 hlim=64 time=0.371 ms
^C
--- f4d488602d4f.local ping6 statistics ---
5 packets transmitted, 5 packets received, 0.0% packet loss
round-trip min/avg/max/std-dev = 0.226/0.310/0.371/0.048 ms
```
but I'm not sure if the ping triggers only the local.host @ 127.0.0.1. But if I quit the OTA-Provider, the above mentioned ping don't succeed. 

Hmmmm, from my OTBR I'm not able to directly ping the OTA-Provider. I try to use the Mesh-Local IPv6 Address (ML-EID --> fd00::10c9:ead4:155f:7086).
```
pi@raspberrypi:~ $ sudo ot-ctl ping fd00::10c9:ead4:155f:7086
1 packets transmitted, 0 packets received. Packet loss = 100.0%.
Done
pi@raspberrypi:~ $ sudo ot-ctl ping fd11:1111:1122:2222:189a:2d36:e093:227a
1 packets transmitted, 0 packets received. Packet loss = 100.0%.
Done
pi@raspberrypi:~ $
```
The ping from the OTBR shell (without executing the ot-ctl service) works
```
pi@raspberrypi:~ $ ping fd00::10c9:ead4:155f:7086
PING fd00::10c9:ead4:155f:7086(fd00::10c9:ead4:155f:7086) 56 data bytes
64 bytes from fd00::10c9:ead4:155f:7086: icmp_seq=1 ttl=255 time=4.81 ms
64 bytes from fd00::10c9:ead4:155f:7086: icmp_seq=2 ttl=255 time=4.24 ms
64 bytes from fd00::10c9:ead4:155f:7086: icmp_seq=3 ttl=255 time=103 ms
64 bytes from fd00::10c9:ead4:155f:7086: icmp_seq=4 ttl=255 time=29.0 ms
^C
--- fd00::10c9:ead4:155f:7086 ping statistics ---
4 packets transmitted, 4 received, 0% packet loss, time 3005ms
rtt min/avg/max/mdev = 4.235/35.347/103.376/40.524 ms
```

Damian-Nordic, 2022-09-14:
Hmm. Not sure if I understand. I'm particularly interested if you are able to ping your OTA requestor (light switch) from your MAC M1 MAX that runs the OTA provider. Could you share output `ot ipaddr` shell command on the light switch, too?

If the ping does not work, it is worth checking if  your MAC M1 AX have routes to the Thread on-mesh addresses via OTBR. I'm not familiar with macs, but OTBR by default sends proper router advertisements, but maybe you need to change something on the mac to accept them.

benjaminnestler, 2022-09-15:
hi @Damian-Nordic,

the ping works

[image]

benjaminnestler, 2022-09-15:
I will try to build the OTA-Provider in an Ubuntu VM on my machine. Maybe something goes wrong during build procedure at my macOS.

benjaminnestler, 2022-09-15:
Hi @Damian-Nordic 

I try to run the OTA-Provider in my Ubuntu VM. The OTA-procedure works! Seems to be an Issue with the OTA-provider app running on a Mac M1 Max (Apple Silicon).

closed 2023-03-23 by stale[bot]