project-chip/connectedhomeip #22626: [SQA]  BINFO-2.2  Shutdown event is Failed in Master branch.

Saravana-kr22, 2022-09-14:
**Issue:**
In BINFO-2.2 step 2 Shutdown event is not working as expected. Chip-tool fails to capture the shut-down function of DUT.

**Steps to Reproduce:**

Start the interactive mode : `./chip-tool interactive start`
Read the shutdown event: `basic subscribe-event shut-down 20 100 1 0`
Then shutdown the DUT.

**Expected behavior :**

> CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0028 Event 0x0000_0001
[1653050528.900202][119367:119372] CHIP:TOO:          Event number: 5
[1653050528.900295][119367:119372] CHIP:TOO:          Priority: Critical
[1653050528.900386][119367:119372] CHIP:TOO:          Timestamp: 289436805
[1653050528.900630][119367:119372] CHIP:TOO:   ShutDown: {
[1653050528.900732][119367:119372] CHIP:TOO:    }
[1653050528.900892][119367:119372] CHIP:DMG: Refresh LivenessCheckTime for 28000 milliseconds with SubscriptionId = 0x27408c83 Peer = 01:0000000000000001

**Actual behavior :**

No event responce
> [1663151639.296874][4354:4359] CHIP:DMG: SubscribeResponse is received
[1663151639.296942][4354:4359] CHIP:DMG: SubscribeResponseMessage =
[1663151639.296989][4354:4359] CHIP:DMG: {
[1663151639.297040][4354:4359] CHIP:DMG: 	SubscriptionId = 0xf18f76cc,
[1663151639.297093][4354:4359] CHIP:DMG: 	MaxInterval = 0x64,
[1663151639.297142][4354:4359] CHIP:DMG: 	InteractionModelRevision = 1
[1663151639.297193][4354:4359] CHIP:DMG: }
[1663151639.297246][4354:4359] CHIP:DMG: Subscription established with SubscriptionID = 0xf18f76cc MinInterval = 20s MaxInterval = 100s Peer = 01:0000000000000001

**System configuration:**

- Sha: 17c5878c79533b0a33b6a025ec884f2b52fd0cdd (master)
- DUT : All-cluster-app ( RPI)
- CNTRL : Chiptool ( RPI)
- Network : Ble-Wifi

**log:**

[binfo-2.2.txt](https://github.com/project-chip/connectedhomeip/files/9566979/binfo-2.2.txt)

mrjerryjohns, 2022-09-14:
You've setup a min interval of 20s, which means that if you emit the shutdown event on the target within 20s of setting up the subscription, it's not going to get delivered as per the spec.

Saravana-kr22, 2022-09-15:
Thank you @mrjerryjohns i understood that ,This Issue is resolved.

closed 2022-09-15 by Saravana-kr22

lisaaaas, 2023-04-06:
Hello, I want to know what shut down the DUT means. I have the same result as you, but I have not solved this problem at present

lisaaaas, 2023-04-06:
> Thank you @mrjerryjohns i understood that ,This Issue is resolved.

Hello, I want to know what shut down the DUT means. I have the same result as you, but I have not solved this problem at present