project-chip/connectedhomeip #22647: [TH] Fixing Manual and automated tests for SVE-2

raju-apple, 2022-09-15:
### Reproduction steps / Feature

```bash
Fixing this https://github.com/CHIP-Specifications/chip-test-scripts/issues/469

and modifying the following TCs

Test_TC_ACE_1_1.yaml
Test_TC_ACL_2_5.yaml
Test_TC_BR_2.yaml
Test_TC_BR_4.yaml
Test_TC_G_3_2.yaml
Test_TC_IDM_2_1.yaml
Test_TC_IDM_6_2.yaml
Test_TC_LVL_2_3.yaml
Test_TC_MOD_1_3.yaml
Test_TC_OO_3_1.yaml
Test_TC_PSCFG_3_1.yaml
Test_TC_PS_3_1.yaml
```

### Platform

raspi

### Type

YAML tested, CI tested

bzbarsky-apple, 2022-09-21:
@raju-apple did https://github.com/project-chip/connectedhomeip/pull/22631 fix this?

raju-apple, 2022-09-22:
Yes

closed 2022-09-22 by raju-apple